DanielMartensson/Open-SAE-J1939 #3: STM32 Issues Compiling and Debugging

Ra13fael, 2022-01-12:
I am using STM32IDE and the STM32L432KC MCU, I am trying to use the J1939 protocol stack to read and talk with a battery. But I am having issues when I am trying to initialize the CAN and my thread crashes on STM32_PLC_CAN_GET_ID_Data.
Am I not using the library correctly??

/* USER CODE BEGIN Header */
/**
  ******************************************************************************
  * @file           : main.c
  * @brief          : Main program body
  ******************************************************************************
/* USER CODE END Header */
/* Includes ------------------------------------------------------------------*/
#include "main.h"

/* Private includes ----------------------------------------------------------*/
/* USER CODE BEGIN Includes */

#include "Hardware.h"
#include <stdlib.h>
#include <stdio.h>

#include "ISO_11783/ISO_11783-7_Application_Layer/Application_Layer.h"
#include "Open_SAE_J1939/Open_SAE_J1939.h"
#include "SAE_J1939/SAE_J1939-71_Application_Layer/Application_Layer.h"
#include "SAE_J1939/SAE_J1939-73_Diagnostics_Layer/Diagnostics_Layer.h"
#include "SAE_J1939/SAE_J1939-81_Network_Management_Layer/Network_Management_Layer.h"

/* USER CODE END Includes */

/* Private variables ---------------------------------------------------------*/
CAN_HandleTypeDef hcan1;

/* Private function prototypes -----------------------------------------------*/
void SystemClock_Config(void);
static void MX_GPIO_Init(void);
static void MX_CAN1_Init(void);

/* USER CODE */
void STM32_PLC_Start_CAN(CAN_HandleTypeDef *hcan, J1939 *j1939);
void STM32_PLC_CAN_Get_ID_Data(uint32_t* ID, uint8_t data[], bool* is_new_message);
static CAN_HandleTypeDef *can_handler;
static J1939 *j1939_handler;
HAL_StatusTypeDef STM32_PLC_CAN_Transmit(uint8_t TxData[], CAN_TxHeaderTypeDef *TxHeader);
static void Create_CAN_Filter(CAN_HandleTypeDef *hcan);
static void Create_CAN_Interrupt(CAN_HandleTypeDef *hcan);
/* USER CODE END */

/**
  * @brief  The application entry point.
  * @retval int
  */
int main(void)
{

  /* MCU Configuration--------------------------------------------------------*/

  /* Reset of all peripherals, Initializes the Flash interface and the Systick. */
  HAL_Init();

  /* Configure the system clock */
  SystemClock_Config();
  /* Initialize all configured peripherals */
  MX_GPIO_Init();
  MX_CAN1_Init();

  /* USER CODE BEGIN WHILE */
  	/* Create our J1939 structure with two ECU */
  	J1939 j1939_1 = {0};
  	J1939 j1939_2 = {0};

  	/* Important to sent all non-address to 0xFF - Else we cannot use ECU address 0x0 */
[... 229 lines cut ...]
  /* USER CODE BEGIN Error_Handler_Debug */
  /* User can add his own implementation to report the HAL error return state */
  __disable_irq();
  while (1)
  {
  }
  /* USER CODE END Error_Handler_Debug */
}

#ifdef  USE_FULL_ASSERT
/**
  * @brief  Reports the name of the source file and the source line number
  *         where the assert_param error has occurred.
  * @param  file: pointer to the source file name
  * @param  line: assert_param error line source number
  * @retval None
  */
void assert_failed(uint8_t *file, uint32_t line)
{
  /* USER CODE BEGIN 6 */
  /* User can add his own implementation to report the file name and line number,
     ex: printf("Wrong parameters value: file %s on line %d\r\n", file, line) */
  /* USER CODE END 6 */
}
#endif /* USE_FULL_ASSERT */

DanielMartensson, 2022-01-14:
Yes. You are using it totaly wrong!

Let's say that you STM32 board is your ECU (Eletrical Controller Unit) and you want to implement Open SAE J1939, then you should use this example: https://github.com/DanielMartensson/Open-SAE-J1939/blob/main/Src/Examples/Open%20SAE%20J1939/Startup.txt.
Use that, then you use the functions you want to communicate with other ECU:s. SAE 1939 is a standard, but in Open SAE J1939 Library, there are also another ISO library for hydraulical valves....if you want to use. Sonceboze(Swiz) Valves are using that standard.

DanielMartensson, 2022-01-14:
By the way! What commando do you want to use?

closed 2022-01-18 by DanielMartensson