DanielMartensson/Open-SAE-J1939 #7: You can use bit fields struct to handle all bit-operations

ShunzDai, 2022-03-01:


DanielMartensson, 2022-03-01:
Yes I know. But will it be easier to write the code?

ShunzDai, 2022-03-01:
very easy, you can see my code https://github.com/ShunzDai/J1939-Protocol/blob/master/src/protocol/j1939_protocol.c

DanielMartensson, 2022-03-01:
> This dramatically reduces the amount of code

Yes. I see. But it does not follows MIRSA C standard. Very good code, you have a lot of skills writing C code at an advanced and professional level. But I have difficulties to see a pattern in the code.

Open SAE J1939 follows the same pattern no matter what type of function it is:
```
Request -> Response -> Read response
```

I want to try to have it simple as possible, even if it results more code.

ShunzDai, 2022-03-01:
sorry I didn't know there was such a standard, it seems like doesn't allow to use malloc and union struct. learn something new lol.

bit fields struct is very usefull, this sould to be irrelevant to the standard

DanielMartensson, 2022-03-01:
> sorry I didn't know there was such a standard, it seems like doesn't allow to use malloc and union struct. learn something new lol.
> 
> bit fields struct is very usefull, this sould to be irrelevant to the standard

That's OK. Your code still works and is excellent. It suits purposes too. =)

Can you post a minimal C example how bit field structs works?

ShunzDai, 2022-03-01:
thanks lol. 

I found this information https://wiki.sei.cmu.edu/confluence/display/c/EXP11-C.+Do+not+make+assumptions+regarding+the+layout+of+structures+with+bit-fields

i think misra c has very high requirements for code stability. using bit field struct, you have to consider big endian and little endian, and cype cast, which misra c does not allow.

DanielMartensson, 2022-03-01:
> 

Yes!
Here I'm casting bytes to a struct, without using bit fields.
https://github.com/DanielMartensson/Open-SAE-J1939/blob/2043ea9ea5a69e7ccd0240d704807bb68d7ef224/Src/Open_SAE_J1939/Startup_ECU.c#L20

It follows the MIRSA C standard.