DanielMartensson/Open-SAE-J1939 #13: Time triggered message transmission

LHSmicius, 2022-07-26:
Super useful stuff, Daniel! Thank you for sharing it.
Just want to make an issue for a new feature - Time triggered message transmission, as many of J1939 messages are not requested, but time triggered. In documentation this parameter is often named as "Repetition Rate" or "Transmission Rate".

As far as I understand, for this feature to work ENUM_PGN_CODES enumerator should be transformed into the data structure with at least these fields:
PGN - same hex numbers as it is now.
Repetition Rate - number representing time in milliseconds.
Next Transmission Time - stores timestamp value when the next transmission should be executed.

Optionally DLC (Data Length Code) can be added to the structure.

Implementation part of this post is only my thoughts, maybe it's not an efficient way to program.

DanielMartensson, 2022-07-27:
> Super useful stuff, Daniel! Thank you for sharing it. Just want to make an issue for a new feature - Time triggered message transmission, as many of J1939 messages are not requested, but time triggered. In documentation this parameter is often named as "Repetition Rate" or "Transmission Rate".
> 
> As far as I understand, for this feature to work ENUM_PGN_CODES enumerator should be transformed into the data structure with at least these fields:
> 
> * PGN - same hex numbers as it is now.
> * Repetition Rate - number representing time in milliseconds.
> * Next Transmission Time - stores timestamp value when the next transmission should be executed.
>   Optionally DLC (Data Length Code) can be added to the structure.
> 
> And also one function for looping all transmit messages is needed, which can be called similarly to Open_SAE_J1939_Listen_For_Messages(J1939* j1939) function (I mean in infinite loop).
> 
> Implementation part of this post is only my thoughts, maybe it's not an efficient way to program.

Hello!

"Time triggered message transmission, as many of J1939 messages are not requested, but time triggered. In documentation this parameter is often named as "Repetition Rate" or "Transmission Rate"."

This software is 100% hardware independent so if you want to have a transmission rate, then you only could add a function to your infinitive loop. I like threads, but threads requries lots of hardware specific code.

"As far as I understand, for this feature to work ENUM_PGN_CODES enumerator should be transformed into the data structure with at least these fields:"

The PGN codes are only common codes for calling a function. If I send you the code 0x00EE00, then you are going to provide me information with the function 0x00EE00.

"And also one function for looping all transmit messages is needed, which can be called similarly to Open_SAE_J1939_Listen_For_Messages(J1939* j1939) function (I mean in infinite loop)."

Yes. This is how I had structure this software. It's a free choice to modify the code and add new PGN functions. Just follow the same structure because this project have a very clear structure and follows the same pattern.

LHSmicius, 2022-08-03:
I'm working on my solution to this issue, but the development doesn't go fast... After I finish I would like to show the idea. I'm not saying that my idea is good, but hope there will be something interesting to think about.

DanielMartensson, 2022-08-03:
> I'm working on my solution to this issue, but the development doesn't go fast... After I finish I would like to show the idea. I'm not saying that my idea is good, but hope there will be something interesting to think about.

Yes. You can show me if you want. I like to see how other solve their problems. :)

closed 2022-08-15 by DanielMartensson