DanielMartensson/Open-SAE-J1939 #25: Acknowledgement与Protocol_Connection_Management有问题

semitimes, 2023-11-06:
下面有两个问题
SAE_J1939_Send_Acknowledgement()
{
    data[0] = control_byte;
    data[1] = group_function_value;														/* The cause of the control byte */
    data[2] = 0xFF;																		/* Reserved */
    data[3] = 0xFF;																		/* Reserved */
    data[4] = j1939->information_this_ECU.this_ECU_address;			/* This source address */
}    
SAE_J1939_Send_Transport_Protocol_Connection_Management()
{
    data[0] = j1939->this_ecu_tp_cm.control_byte;
    data[1] = j1939->this_ecu_tp_cm.total_message_size;
    data[2] = j1939->this_ecu_tp_cm.total_message_size >> 8;
    data[3] = j1939->this_ecu_tp_cm.number_of_packages;
}

DanielMartensson, 2023-11-13:
?

semitimes, 2023-11-14:
maybe i was wrong
i want to find this function (TP.CM_CTS)，but i run in this function（SAE_J1939_Send_Acknowledgement()）

semitimes, 2023-11-14:
i want to find this function (TP.CM_EndofMsgAck)，but i run in this function（SAE_J1939_Send_Acknowledgement()）

semitimes, 2023-11-14:
maybe i was wrong
i want to find this function (TP.CM_CTS)，but i run in this function（SAE_J1939_Send_Transport_Protocol_Connection_Management()）

DanielMartensson, 2023-11-14:
> maybe i was wrong i want to find this function (TP.CM_CTS)，but i run in this function（SAE_J1939_Send_Transport_Protocol_Connection_Management()）

TP.CM_CTS is not a function, is an enum in the C-language. CTS stands for Clear To Send.

GIT-LEE4, 2023-11-15:
文件: Transport_Protocol_Connection_Management 
函数: SAE_J1939_Read_Transport_Protocol_Connection_Management
 /* Check if we got the Request To Send control byte - We need to answer with CTS - Clear To Send */
if(j1939->from_other_ecu_tp_cm.control_byte == CONTROL_BYTE_TP_CM_RTS)
{
    j1939->this_ecu_tp_cm = j1939->from_other_ecu_tp_cm; 			/* Copy - We need to have the same data */
    j1939->this_ecu_tp_cm.control_byte = CONTROL_BYTE_TP_CM_CTS;	/* We only need to change the control byte from RTS to CTS */
    SAE_J1939_Send_Transport_Protocol_Connection_Management(j1939, SA);// 这个地方虽然发送的control_byte 是CTS，实际函数内发送的内容却是按照RTS来组织的
}
此处按照你的思路应该是回应CTS信息，向发送方请求指定的数据包数量和数据包开始的索引
但是看你对这个函数的实现和其他地方对这个函数的调用情况，这个函数似乎实际上是在发送RTS信息
ENUM_J1939_STATUS_CODES SAE_J1939_Send_Transport_Protocol_Connection_Management(J1939 *j1939, uint8_t DA) {
	uint32_t ID = (0x1CEC << 16) | (DA << 8) | j1939->information_this_ECU.this_ECU_address;
	uint8_t data[8] = { 0 };
	data[0] = j1939->this_ecu_tp_cm.control_byte;
	data[1] = j1939->this_ecu_tp_cm.total_message_size;  // 这里实际上存储了整个消息的字节数量
	data[2] = j1939->this_ecu_tp_cm.total_message_size >> 8;
	data[3] = j1939->this_ecu_tp_cm.number_of_packages; // 这里实际上存储了消息的包数量
	data[4] = 0xFF; 															/* Reserved */
	data[5] = j1939->this_ecu_tp_cm.PGN_of_the_packeted_message;
	data[6] = j1939->this_ecu_tp_cm.PGN_of_the_packeted_message >> 8;
	data[7] = j1939->this_ecu_tp_cm.PGN_of_the_packeted_message >> 16;
	return CAN_Send_Message(ID, data);
}

DanielMartensson, 2023-11-15:
?

GIT-LEE4, 2023-11-15:
> ？

function: SAE_J1939_Send_Transport_Protocol_Connection_Management
The internal code of this function implements the sending of RTS information.
When the function “SAE_J1939_Read_Transport_Protocol_Connection_Management” calls it, 
it is expected to send CTS information.

DanielMartensson, 2023-11-15:
Can you be more specific? 
What is the question? What do you want to achieve ?

GIT-LEE4, 2023-11-16:
> Can you be more specific? What is the question? What do you want to achieve ?

[image]
[image]

DanielMartensson, 2023-11-16:
I have made an explanation inside the ReadMe.md file.

DanielMartensson, 2023-11-17:
The reason why I shift the RTS to CTS and send the same message, is because there are no master or slave in the SAE J1939 protocol.

closed 2023-11-17 by DanielMartensson