DanielMartensson/Open-SAE-J1939 #28: Problem building Project

Navjot0005, 2023-12-16:
Hi @DanielMartensson thanks for the library you provided.
I am working making a charger to communicate with BMS using J1939.
Here I could not build the project.
I follow the instructions you given.
I made new project empty in CubeIDE>>then I renamed the source folder and then copy and paste in project folder.
But when I build project it shows some kind of fatal errors.
I copied the main.text file into the main.c that is my project main file.
following the code I copied
Please check it and let me know.
Waiting....
/*
 * Main.c
 *
 *  Created on: 16 juli 2021
 *      Author: Daniel Mårtensson
 */

#include <stdio.h>

 /* Include Open SAE J1939 */
#include "Open_SAE_J1939/Open_SAE_J1939.h"

/* Include ISO 11783 */
#include "ISO_11783/ISO_11783-7_Application_Layer/Application_Layer.h"

void Callback_Function_Send(uint32_t ID, uint8_t DLC, uint8_t data[]) {
	/* Apply your transmit layer here, e.g:
	 * uint32_t TxMailbox;
	 * static CAN_HandleTypeDef can_handler;
	 * This function transmit ID, DLC and data[] as the CAN-message.
	 * HardWareLayerCAN_TX(&can_handler, ID, DLC, data, &TxMailbox);
	 *
	 * You can use TCP/IP, USB, CAN etc. as hardware layers for SAE J1939
	 */
}

void Callback_Function_Read(uint32_t* ID, uint8_t data[], bool* is_new_data) {
	/* Apply your receive layer here, e.g:
	 * CAN_RxHeaderTypeDef rxHeader = {0};
	 * static CAN_HandleTypeDef can_handler;
	 * This function read CAN RX and give the data to ID and data[] as the CAN-message.
	 * if (HardWareLayerCAN_RX(can_handler, &rxHeader, ID, data) == STATUS_OK){
	 *	*is_new_data = true;
	 * }
	 *
	 * You can use TCP/IP, USB, CAN etc. as hardware layers for SAE J1939
	 */
}

/* This function reads the CAN traffic */
void Callback_Function_Traffic(uint32_t ID, uint8_t DLC, uint8_t data[], bool is_TX) {
	/* Print if it is TX or RX */
	printf("%s\t", is_TX ? "TX" : "RX");

	/* Print ID as hex */
	printf("%08X\t", ID);

	/* Print the data */
	uint8_t i;
	for (i = 0U; i < DLC; i++) {
		printf("%X\t", data[i]);
	}

	/* Print the non-data */
	for (i = DLC; i < 8U; i++) {
		printf("%X\t", 0U);
	}

	/* New line */
	printf("\n");
}

int main() {

	/* Create our J1939 structure */
	J1939 j1939 = { 0 };

	/*
	 * Callbacks can be used if you want to pass a specific CAN-function into the hardware layer.
	 * All you need to do is to enable INTERNAL_CALLLBACK inside hardware.h
	 * If you don't want to have the traffic callback, just set the argument as NULL.
	 * If you don't want any callback at all, you can write your own hardware layer by selecting a specific procesor choice at hardware.h
	 */
	CAN_Set_Callback_Functions(Callback_Function_Send, Callback_Function_Read, Callback_Function_Traffic);

	/* Load your ECU information */
	Open_SAE_J1939_Startup_ECU(&j1939);

	/* SAE J1939 process */
	bool run = true;
	while (run) {
		/* Read incoming messages */
		Open_SAE_J1939_Listen_For_Messages(&j1939);

		/* Your application code here */

	}

	/* Save your ECU information */
	Open_SAE_J1939_Closedown_ECU(&j1939);

	return 0;
}

DanielMartensson, 2023-12-16:
What error did you get ?

Navjot0005, 2023-12-16:
[image]
Hi 
Thanks for your reply @DanielMartensson .
Please see the attached screen short for the same.

DanielMartensson, 2023-12-16:
@Navjot0005 Ah, I see. You have forgot the .c and .h files.

Navjot0005, 2023-12-16:
where are those files placed???

Navjot0005, 2023-12-16:
@DanielMartensson I am able to build the code now.
Can you please tell me where shall I give the identifier id dlc means where I get I make a frame so that it can catch associated information.
and after sending the frame where shall I get the msg how it will parse??
Please help me for this.

DanielMartensson, 2023-12-16:
> @DanielMartensson I am able to build the code now. Can you please tell me where shall I give the identifier id dlc means where I get I make a frame so that it can catch associated information. and after sending the frame where shall I get the msg how it will parse?? Please help me for this.

@Navjot0005 
The ID is defined in your protocol, same as the DLC. You don't need to focus on that. 

Give your ECU:s two different address e.g 0x20 and 0x21. Then you can communicate with each other.

Make sure you are calling the read messages function all the time.

Navjot0005, 2023-12-16:
@DanielMartensson In example I check there are four folders Named as 1. "Hardware", 2. "ISO11783", 3. Open_SAE_J1939, 4. SAE_J1939.

Now please tell me which folder name shall I open and use ???
And why files placed in these four folders are in .txt format??

DanielMartensson, 2023-12-16:
> @DanielMartensson In example I check there are four folders Named as 1. "Hardware", 2. "ISO11783", 3. Open_SAE_J1939, 4. SAE_J1939.
> 
> Now please tell me which folder name shall I open and use ??? And why files placed in these four folders are in .txt format??

 @Navjot0005 
The `.txt` files are just examples you just copy and paste and run. 

Try to run this example.
https://github.com/DanielMartensson/Open-SAE-J1939/blob/main/Src/Examples/SAE%20J1939/ECU%20Identification.txt

Navjot0005, 2023-12-16:
@DanielMartensson  shall I copy this file into main.c??
What this example for ??
Please tell If I flash in the board what it will do??

DanielMartensson, 2023-12-16:
> @DanielMartensson shall I copy this file into main.c?? What this example for ?? Please tell If I flash in the board what it will do??

Do a offline simulation first. With a regular Windows compiler or so.
Then run make sure you understand how to use Open SAE J1939.

After you understand how to use Open SAE J1939, copy over your code to your STM32.

Navjot0005, 2023-12-16:
@DanielMartensson OK...Do I need to activate processor in hardware.h file??
Got it.. Can you please tell which simulator shall I use on PC??

DanielMartensson, 2023-12-16:
> @DanielMartensson OK...Do I need to activate processor in hardware.h file?? Got it.. Can you please tell which simulator shall I use on PC??

You can use Visual Studio, Eclipse, JetBrains etc. GCC-compiler or MSVC. 
When you are running Open SAE J1939, just use internal in hardware.h

Navjot0005, 2023-12-16:
@DanielMartensson Can I simulate in CUBEIDE???

DanielMartensson, 2023-12-16:
> @DanielMartensson Can I simulate in CUBEIDE???

Yes, but it's much easier if you doing that in Windows.

Navjot0005, 2023-12-16:
@DanielMartensson  ok....
Do I need STM32F board for simulation???
Or will it work without any controller board??

DanielMartensson, 2023-12-16:
> @DanielMartensson ok.... Do I need STM32F board for simulation??? Or will it work without any controller board??

Try to make it work without a controller first. Learn how to use the library.

Navjot0005, 2023-12-16:
> > @DanielMartensson ok.... Do I need STM32F board for simulation??? Or will it work without any controller board??
> 
> Try to make it work without a controller first. Learn how to use the library.

@DanielMartensson any document I can refer to how to simulate the example code and uses of library??

DanielMartensson, 2023-12-16:
> > > @DanielMartensson ok.... Do I need STM32F board for simulation??? Or will it work without any controller board??
> > 
> > 
> > Try to make it work without a controller first. Learn how to use the library.
> 
> @DanielMartensson any document I can refer to how to simulate the example code and uses of library??

Just copy the `.txt` files into your `main.c` and press run.

Navjot0005, 2023-12-16:
@DanielMartensson I run the 

> @Navjot0005 The `.txt` files are just examples you just copy and paste and run.
>
> Try to run this example. [...]

@DanielMartensson I tried this example and build it.
Now when I run the code I am getting a window pop up saying that connect ST-Link first..

DanielMartensson, 2023-12-16:
> @DanielMartensson I run the
>
>
> @DanielMartensson I tried this example and build it. Now when I run the code I am getting a window pop up saying that connect ST-Link first..

Use Visual studio. Don't use ST, microcontroller etc right now. First run the Open SAE J1939 on Windows.

Navjot0005, 2023-12-16:
@DanielMartensson Ok. Let me try with Visual Studio and will get back to you soon.
Thanks so much for the help. It really appreciated.!!!

DanielMartensson, 2023-12-16:
> @DanielMartensson Ok. Let me try with Visual Studio and will get back to you soon. Thanks so much for the help. It really appreciated.!!!

I recommend visual studio community. Not Visual Studio Code.

Navjot0005, 2023-12-16:
@DanielMartensson Is it a simulation software??
Any link or video for software??
Because when I googled it shows Visual Studio Code..

DanielMartensson, 2023-12-16:
> @DanielMartensson Is it a simulation software?? Any link or video for software?? Because when I googled it shows Visual Studio Code..

Visual Studio Community has been here for over 20 years. 
Use Visual Studio Community. It's from Microsoft.

Navjot0005, 2023-12-16:
[image]
 @DanielMartensson ok..I have downloaded the VS community installer. So now shall I open project in VS community??

DanielMartensson, 2023-12-16:
Yes. Open the project file

closed 2023-12-22 by DanielMartensson