DanielMartensson/Open-SAE-J1939 #29: how Read multi packet of other ECU

payapars, 2023-12-16:
thk for  for open source stack.

1-i want read multi packect of other ecu .but i did understand in catalog and source code.please hlep me.
2-i think this stack needed RTOS for manegment send and timeout for  this stack .

DanielMartensson, 2023-12-17:
What do you mean ?

payapars, 2023-12-17:
i edited my answer!

DanielMartensson, 2023-12-17:
> i edited my answer!

Sorry, I dont understand. Use the read messages function in your RTOS thread.

payapars, 2023-12-17:
this my exp:

`ej1939MasterReqErrCode WriteMultiBufferData(uint8_t ucSndAddr,uint32_t PNG,uint8_t *Data,uint16_t datalen,LONG lTimeOut){
	
	
    ej1939MasterReqErrCode    eErrStatus = J1939_MRE_NO_ERR;
   // uint16_t T
    if ( ucSndAddr > J1939_MASTER_TOTAL_SLAVE_NUM ||datalen>MAX_TP_DT||datalen==NULL) eErrStatus = J1939_MRE_ILL_ARG;
    else if ( xj1939MasterRunResTake( lTimeOut ) == FALSE ) eErrStatus = J1939_MRE_MASTER_BUSY;
    else
    {
			  j1939.from_other_ecu_tp_cm.PGN_of_the_packeted_message=PNG;
		   	j1939.from_other_ecu_tp_cm.control_byte=CONTROL_BYTE_TP_CM_RTS;
			  j1939.this_ecu_tp_cm.number_of_packages=datalen/7;
				/* set number of packages, where each package will transmit up to 7 bytes */
				if (datalen % 7 > 0){
					j1939.this_ecu_tp_cm.number_of_packages+=1;	/* add extra frame if data rolls over */
				}
		    j1939.from_other_ecu_tp_cm.total_message_size=datalen;
				
				
				memcpy(&j1939.data[0],Data,datalen);
				
       SAE_J1939_Send_Transport_Protocol_Connection_Management(&j1939, ucSndAddr);
				
		   eErrStatus = ej1939MasterWaitRequestFinish(lTimeOut );
    }
    return eErrStatus;

}`

and function wait:

`ej1939MasterReqErrCode ej1939MasterWaitRequestFinish(long time_out ) {
  ej1939MasterReqErrCode    eErrStatus_Req = J1939_MRE_NO_ERR;
	uint32_t eErrStatus=J1939_MRE_NO_ERR;
          	j1939_LEDRxCpltCallback(RESET);
						j1939_LEDTxCpltCallback(SET);	
 	eErrStatus=	osEventFlagsWait (J1939_evt_id,
								EV_MASTER_PROCESS_SUCESS  
							| EV_MASTER_ERROR_RESPOND_TIMEOUT
							| EV_MASTER_ERROR_RECEIVE_DATA
							| EV_MASTER_ERROR_EXECUTE_FUNCTION,
								osFlagsWaitAny, time_out);
	
    switch (eErrStatus)
    {
					case EV_MASTER_PROCESS_SUCESS:
					{
					    eErrStatus_Req=	J1939_MRE_NO_ERR;	
							break;
					}
					case EV_MASTER_ERROR_RESPOND_TIMEOUT:
					{
							eErrStatus_Req = J1939_MRE_TIMEDOUT;
							break;
					}
					case EV_MASTER_ERROR_RECEIVE_DATA:
					{
							eErrStatus_Req = J1939_MRE_REV_DATA;
							break;
					}
					case EV_MASTER_ERROR_EXECUTE_FUNCTION:
					{
							eErrStatus_Req = J1939_MRE_EXE_FUN;
							break;
					}
					default:{
					  eErrStatus_Req =	J1939_MRE_TIMEDOUT;
						break;
[... 62 lines cut ...]
			SAE_J1939_Read_Response_Request_Component_Identification(j1939, SA, data);
		}else if (id0 == 0x0C && id1 == 0xFE && DA >= 0x10 && DA <= 0x1F){
			ISO_11783_Read_Response_Request_Auxiliary_Estimated_Flow(j1939, SA, DA & 0xF, data);				/* DA & 0xF = Valve number. Total 16 valves from 0 to 15 */
		}else if (id0 == 0x0C && id1 == 0xC6 && DA == j1939->information_this_ECU.this_ECU_address){
			ISO_11783_Read_Response_Request_General_Purpose_Valve_Estimated_Flow(j1939, SA, data);
		}else if (id0 == 0x0C && id1 == 0xFF && DA >= 0x20 && DA <= 0x2F){
			ISO_11783_Read_Response_Request_Auxiliary_Valve_Measured_Position(j1939, SA, DA & 0xF, data); 		/* DA & 0xF = Valve number. Total 16 valves from 0 to 15 */

		/* Read command from other ECU */
		}else if (id0 == 0x0C && id1 == 0xFE && DA >= 0x30 && DA <= 0x3F){
			ISO_11783_Read_Auxiliary_Valve_Command(j1939, SA, DA & 0xF, data); 									/* DA & 0xF = Valve number. Total 16 valves from 0 to 15 */
		}else if (id0 == 0x0C && id1 == 0xC4 && DA == j1939->information_this_ECU.this_ECU_address){
			ISO_11783_Read_General_Purpose_Valve_Command(j1939, SA, data);										/* General Purpose Valve Command have only one valve */
		}else if (id0 == 0x0 && id1 == 0x2 && (DA == j1939->information_this_ECU.this_ECU_address || DA == 0xFF)){
			SAE_J1939_Read_Address_Delete(j1939, data);															/* Not a SAE J1939 standard */
		}else if (id0 == 0x1D && id1 == 0xEB && DA == j1939->information_this_ECU.this_ECU_address){
			SAE_J1939_Read_Transport_Protocol_Data_Transfer(j1939, SA, data);
		}
		else{
			is_new_message = false;																				/* The message was not meant for this ECU */
		}
		/* Add more else if statement here */
	}
	return is_new_message;
}`

DanielMartensson, 2023-12-18:
What do you want to achieve?

payapars, 2023-12-20:
In your code to send data if it is in master mode, there is no waiting time! Is this correct?

DanielMartensson, 2023-12-21:
> In your code to send data if it is in master mode, there is no waiting time! Is this correct?

@payapars you can add your own delay at hardware setup of this library

DanielMartensson, 2023-12-21:
See here.

https://github.com/DanielMartensson/Open-SAE-J1939/blob/677c5839992d37b8bb032472abc9ab7fe5559723/Src/Hardware/CAN_Transmit_Receive.c#L183

If you want to add a call back, then send me a pull request

closed 2023-12-24 by DanielMartensson