DanielMartensson/Open-SAE-J1939 #30: Charger interfacing with BMS using CAN

Navjot0005, 2023-12-24:
hii...@DanielMartensson
Hope you are good..
I am here to know that can your library be used for communicating 2Wheeler charger with BMS??

DanielMartensson, 2023-12-24:
> hii...@DanielMartensson
> Hope you are good..
> I am here to know that can your library be used for communicating 2Wheeler charger with BMS??

You can use DM16 message.

Navjot0005, 2023-12-24:
> > hii...@DanielMartensson
> > Hope you are good..
> > I am here to know that can your library be used for communicating 2Wheeler charger with BMS??
> 
> You can use DM16 message.

Where it is located??

DanielMartensson, 2023-12-24:
> > > hii...@DanielMartensson
> > > Hope you are good..
> > > I am here to know that can your library be used for communicating 2Wheeler charger with BMS??
> > 
> > You can use DM16 message.
> 
> Where it is located??

First you send a DM14 request, then you want for a DM15 response, followed by a DM16 data transfer.

DM14 is a message about asking other ECU about location and amount of data. DM15 is the response of access of the data. Then DM16 is followed by DM15.

Work with some examples.

Navjot0005, 2023-12-24:
ok let me see..
Thanks for the reply..

Navjot0005, 2023-12-24:
Hi @DanielMartensson
 I can't find the PGN:- 0x00EF00(61184) in the Enum_PGN.h this is used for time sync.

DanielMartensson, 2023-12-24:
> Hi @DanielMartensson
>  I can't find the PGN:- 0x00EF00(61184) in the Enum_PGN.h this is used for time sync.

Do you have the protocol?

Navjot0005, 2023-12-24:
> > Hi @DanielMartensson
> > I can't find the PGN:- 0x00EF00(61184) in the Enum_PGN.h this is used for time sync.
> 
> Do you have the protocol?

What does it mean??

DanielMartensson, 2023-12-24:
> > > Hi @DanielMartensson
> > > I can't find the PGN:- 0x00EF00(61184) in the Enum_PGN.h this is used for time sync.
> > 
> > Do you have the protocol?
> 
> What does it mean??

Every PGN is a code for a function. When you are sending a request (DLC=3) with the PGN number, you are going to get a response back about that PGN number.

Navjot0005, 2023-12-24:
> Every PGN is a code for a function. When you are sending a request (DLC=3) with the PGN number, you are going to get a response back about that PGN number.

ok.. Actually  got a document from BMS company 
[iVec Protocol .pdf](https://github.com/DanielMartensson/Open-SAE-J1939/files/13762480/iVec.Protocol.pdf)
 PLease check this document and let me know  how I can start.. Thanks for the reply..!!

DanielMartensson, 2023-12-24:
> ok.. Actually  got a document from BMS company
> [iVec Protocol .pdf](https://github.com/DanielMartensson/Open-SAE-J1939/files/13762480/iVec.Protocol.pdf)
>  PLease check this document and let me know [...]

Yes. This can be implemented.
The protocol is very well explained with pictures etc.

First thing first.
Study the data frame of that PGN code. 
Study the request frame and the acknowledge frame.

Then we can start coding. You will then see how easy that is to implement code into this library.

But study the data you want to send and receive. Begin with the simplest command first.

Navjot0005, 2023-12-24:
Thats great @DanielMartensson ...
Thanks for the support...do you also do the consultancy ??? 
Firstly I just want to write a code for sync with BMS because BMS can only start sending msgs whn it sync.
Can you please tell me how I can do this?

DanielMartensson, 2023-12-24:
> Thats great @DanielMartensson ...
> Thanks for the support...do you also do the consultancy ??? 
> Firstly I just want to write a code for sync with BMS because BMS can only start sending msgs whn it sync.
> Can you please tell me how I can do this?

Ok. We begin there.

You have an ECU unit and sending out messages and you want to read that. Does it broadcast (DA = 0xFF) or send to specific adress ?

Navjot0005, 2023-12-24:
> Ok. We begin there.
>
> You have an ECU unit and sending out messages and you want to read that. Does it broadcast (DA = 0xFF) or send to specific adress ?

Yes..I have an ECU unit that broadcast messages in Single packet messages and Multi-Packet messages..
It also send Payload for example Each cell voltage of battery, current,SoC,SoH etc...

DanielMartensson, 2023-12-24:
> Yes..I have an ECU unit that broadcast messages in Single packet messages and Multi-Packet messages..
> It also send Payload for example Each cell voltage of battery, current,SoC,SoH etc...

Okej.

Collect the data frame for each message, also the multi package. Then it's easy to code.

Also, search for the ISO standard for what this protocol belongs to or if it's pure SAE J1939-21 or 77.

Navjot0005, 2023-12-24:
How I can collect the data frame ??
Can I read BMS data using any CAN tool??
Or is there any way to read messages coming from BMS??

DanielMartensson, 2023-12-25:
@Navjot0005 Hi!

I have added the PGN 0x00EF00 now and identify what is.
The PGN 0x00EF00 is Property A, which means it contans manufacturer specific data.

All you need to do now is to work on request, acknowledgement and how to read the response

Navjot0005, 2023-12-25:
@DanielMartensson oo great sir then. I got the rough idea of protocol now, the remaining point is the only **timestamp for sync** with BMS..I am studying it also....

DanielMartensson, 2023-12-25:
> @DanielMartensson oo great sir then. I got the rough idea of protocol now, the remaining point is the only **timestamp for sync** with BMS..I am studying it also....

@Navjot0005 

Done. You have your Proprietary A functionality now. Notice that you need to build your own product specificed library that can read the struct of proprietary A. 

Run the Proprietary A.txt example in this project.

Navjot0005, 2023-12-25:
@DanielMartensson...ok Downloading whole project again..

DanielMartensson, 2023-12-25:
> @DanielMartensson...ok Downloading whole project again..

You could use pull command. Then you will get the updated files. 

So now you need to learn how to read the data from Proprietary A and pass around it between your ECU:s.
Then you create your own manufacturer specific layer that can read Proprietary A in your own way.

Notice that the data inside Proprietary A is just a `uint8_t ` array of `N` length. 

If you want, send me a pull request of your manufacture specific layer. Create a folder between Open SAE J1939 folder and label it as `Manufacture_Specific`, just as I did with the folder [ISO_11783](https://github.com/DanielMartensson/Open-SAE-J1939/tree/main/Src/ISO_11783).

`Manufacture_Specific/Vecmocon`. Then you using the SAE J1939 functions inside of your own .C code. 
Also! Make `enums` useful too. See How I structured the ISO_11783 functionality and mapping.

Navjot0005, 2023-12-25:
@DanielMartensson for that I have to make a different repo or shall I add in your existing one??

DanielMartensson, 2023-12-25:
> @DanielMartensson for that I have to make a different repo or shall I add in your existing one??

Yes. Clone this repository and work with the cloned one. Then you create your .c code for your specific device, that's using Open SAE J1939 functionalities. 

Then send a pull request back. Make sure you are following ANSI C standard and not using C99. Use enums, typedef, pointers, const, functions etc. That will result a very professional and clean code.

Navjot0005, 2023-12-25:
> Yes. Clone this repository and work with the cloned one. Then you create your .c code for your specific device, that's using Open SAE J1939 functionalities.
>
> Then send a pull request back. Make sure [...]

Okk...Can I add remote repo in ST Cube IDE??

DanielMartensson, 2023-12-25:
> Okk...Can I add remote repo in ST Cube IDE??

@Navjot0005 
STM32CubeIDE have built in Git and also GCC.

Navjot0005, 2023-12-25:
ok I dint use this before let me see @DanielMartensson....back soon

DanielMartensson, 2023-12-25:
> ok I dint use this before let me see @DanielMartensson....back soon

@Navjot0005  First you write some basic .c code for your application, try it, if it works good, add it into open sae j1939 library.

Navjot0005, 2023-12-25:
ok @DanielMartensson.. for now I am going to make a project in which I will use two STM32F1 series controller and try to transfer data one will send and other will receive.
Will it be good to start?
Also Can I have your email id?

DanielMartensson, 2023-12-25:
> ok @DanielMartensson.. for now I am going to make a project in which I will use two STM32F1 series controller and try to transfer data one will send and other will receive. Will it be good to start? Also Can I have your email id?

Create a project with your STM32F1, then copy over Open SAE J1939 project to your project, and make it work with the basic commands. Then you working on your manufacturer specific layer for your device in a separate folder inside the Open SAE J1939 project. When that's working, send that pull request to this repository.

That will be a good start.

Navjot0005, 2023-12-25:
@DanielMartensson Will

> > ok @DanielMartensson.. for now I am going to make a project in which I will use two STM32F1 series controller and try to transfer data one will send and other will receive. Will it be good to start? Also Can I have your email id?
> 
> @Navjot0005
> 
> Create a project with your STM32F1, then copy over Open SAE J1939 project to your project, and make it work with the basic commands. Then you working on your manufacturer specific layer for your device in a separate folder inside the Open SAE J1939 project. When that's working, send that pull request to this repository.
> 
> That will be a good start.
> 
> Example: https://github.com/DanielMartensson/STM32-PLC/tree/main/STM32%20PLC%20for%20GoobySoft/Core/Src/STM32%20PLC/Open%20SAE%20J1939

What exactly this code does???

Navjot0005, 2023-12-25:
is this also a J1939 based???

DanielMartensson, 2023-12-25:
> is this also a J1939 based???

@Navjot0005 This code is the Open SAE J1939. I just copied over it to a STM32 project. Ctrl+c and then Ctrl+v. Open SAE J1939 does not take account of hardware. It's platform independent.

closed 2023-12-25 by DanielMartensson