DanielMartensson/Open-SAE-J1939 #32: How to write in FLASH or EEPROM using DM 16 DM 15 DM 14

payapars, 2023-12-30:
How to write in FLASH or EEPROM using DM 16 DM 15 DM 14. I didn't see anything in the codes, can you explain?

I understood reading but not writing!

DanielMartensson, 2023-12-30:
> How to write in FLASH or EEPROM using DM 16 DM 15 DM 14. I didn't see anything in the codes, can you explain?
> 
> I understood reading but not writing!

DM14 is a memory request and DM15 is a memory access response, followed by DM16, if access is granted.

See the document of Open SAE J1939 I've posted.

payapars, 2023-12-30:
I don't see any way to write in FLASH? Please help me more!

DanielMartensson, 2023-12-30:
> I don't see any way to write in FLASH? Please help me more!

You need to write your own hardware layer for your access to the EEPROM. 

Assume that you got two ECUs - ECU1 and ECU2.

ECU1 send a DM14 request to ECU2 and ask ECU2 for specific data located in ECU2. ECU2 then answer back with a response, called DM15, if the access is granted or not. If it is granted, the DM15 message will be followed by a DM16 message, which contains the data.

payapars, 2023-12-30:
> Have you run the examples?

not yet!

[image]

Is this my mentality of writing function correct?

DanielMartensson, 2023-12-31:
Hi. No, the DM15 and DM16 comes from the ECU.

payapars, 2023-12-31:
thanks

> Hi. No, the DM15 and DM16 comes from the ECU.

from ECU1?

DanielMartensson, 2024-01-01:
> thanks
> 
> > Hi. No, the DM15 and DM16 comes from the ECU.
> 
> from the ECU1?

No the ECU2

payapars, 2024-01-01:
I did not understand! Can you edit the posted figure to write in ECU 2? thank you

DanielMartensson, 2024-01-01:
> I did not understand! Can you edit the posted figure to write in ECU 2? thank you

Have you read the document of this library ? It will explain how DM14 to DM16 will be used.

payapars, 2024-01-01:
```
ENUM_J1939_STATUS_CODES SAE_J1939_Read_Request_DM14(J1939 *j1939, uint8_t DA, uint8_t data[]) {
	/* These are information the ECU want to have */
	uint16_t number_of_requested_bytes = ((data[1] & 0b11100000) << 3) | data[0];
	uint8_t pointer_type = (data[1] >> 4) & 0b0001;
	uint8_t command = (data[1] >> 1) & 0b0000111;
	uint32_t pointer = (data[4] << 16) | (data[3] << 8) | data[2];
	uint8_t pointer_extension = data[5];
	uint16_t key = (data[7] << 8) | data[6];

	/* Load up the amount of bytes we want to send via DM16 */
	uint8_t raw_binary_data[MAX_TP_DT];

	/* Here we ask the flash, eeprom or ram and use pointers */
	FLASH_EEPROM_RAM_Memory(&number_of_requested_bytes, pointer_type, &command, &pointer, &pointer_extension, &key, raw_binary_data);

	/* Prepare a DM15 PGN: 0x00D800 response back to DA */
	uint16_t number_of_allowed_bytes = number_of_requested_bytes;
	uint8_t status = command;
	uint32_t EDC_parameter = pointer;
	uint8_t EDCP_extention = pointer_extension;
	uint16_t seed = key;
	status = SAE_J1939_Send_Response_DM15(j1939, DA, number_of_allowed_bytes, status, EDC_parameter, EDCP_extention, seed);

	/* Check if our message was OK - Send DM16 binary data transfer */
	if(status == STATUS_DM15_PROCEED) {
		if(SAE_J1939_Send_Binary_Data_Transfer_DM16(j1939, DA, number_of_allowed_bytes, raw_binary_data) == STATUS_SEND_OK){
			status = SAE_J1939_Send_Response_DM15(j1939, DA, number_of_allowed_bytes, STATUS_DM15_OPERATION_COMPLETED, EDC_parameter, EDCP_extention, seed);
		}else{
			status = SAE_J1939_Send_Response_DM15(j1939, DA, number_of_allowed_bytes, STATUS_DM15_OPERATION_FAILED, EDC_parameter, EDCP_extention, seed);
		}
	}
	return status;

}

```
If you mean this  code, this only works for download! This is my take, because
There is no array to receive, send from ECU 1 of FLASH_EEPROM_RAM_Memory function.
It is for reading only.

DanielMartensson, 2024-01-02:
> ```
> ENUM_J1939_STATUS_CODES SAE_J1939_Read_Request_DM14(J1939 *j1939, uint8_t DA, uint8_t data[]) {
> 	/* These are information the ECU want to have */
> 	uint16_t number_of_requested_bytes = ((data[1] [...]

Have you configure your hardware layer too?

payapars, 2024-01-02:
yes.

```

void FLASH_EEPROM_RAM_Memory(uint16_t *number_of_requested_bytes, uint8_t pointer_type, uint8_t *command, uint32_t *pointer, uint8_t *pointer_extension, uint16_t *key, uint8_t *raw_binary_data){

#if PROCESSOR_CHOICE == STM32
		FLASH_StatusTypeDef error;
	if(*command==COMMAND_DM14_READ)
      error=	sFLASH_Read_Main(raw_binary_data,*pointer, *number_of_requested_bytes);
	else if(*command==COMMAND_DM14_WRITE)
		error=	sFLASH_Write_Main(raw_binary_data,*pointer, *number_of_requested_bytes);
	if(error==FLASH_OK)
	{
	  *command = STATUS_DM15_PROCEED;
		*number_of_requested_bytes=*number_of_requested_bytes;
		 return;
	}else if(error==FLASH_ERROR){
		*number_of_requested_bytes=0;
	*command=	STATUS_DM15_OPERATION_FAILED;
	}else if(error==FLASH_BUSY){
		*number_of_requested_bytes=0;
		 *command = STATUS_DM15_BUSY;
	}else{
		*number_of_requested_bytes=0;
	  *command=	STATUS_DM15_OPERATION_FAILED;
	}
	/* Implement your memory handler function for the STM32 platform */

#endif
}

```

DanielMartensson, 2024-01-02:
Good! Does it work when your ECU1 send a DM14 request to ECU2 and ECU2 send back the DM15 response, followed by DM16 (if access is granted)?

payapars, 2024-01-02:
I haven't tested it yet, but I think it works fine for Read, but I don't think it works for Write. I will test in a few days.

DanielMartensson, 2024-01-03:
> > Good! Does it work when your ECU1 send a DM14 request to ECU2 and ECU2 send back the DM15 response, followed by DM16 (if access is granted)?
> 
> I haven't tested it yet, but I think it works fine for Read, but I don't think it works for Write. I will test it in a few days. Have you tried it for WRITE?

@payapars 
I have not tried your code.  
Also, as I remember DM14 is a memory request from another ECU. E.g read only mode.

I haven't found any standard or frame that allows a write request.

payapars, 2024-01-03:
J1939/73 Application Layer – Diagnostics
This document defines functions and messages for accessing diagnostic and calibration data. There is a number of predefined Diagnostic Messages (DM) used for:

- Reading and  **writing** to ECU memory
- Reporting diagnostic information when running
- Identification of lamp status
- Reading and clearing Diagnostic Trouble Codes (DTCs)
- Start/stop broadcast DMs

Newer versions of this document include additional information regarding the interpretation of Diagnostic Trouble Codes (DTC). The DTC is a 32 bit identifier inherited from J1587 and consists of 4 elements: Suspect Parameter Number (SPN), Failure Mode Identifier (FMI), Occurrence Count (OC) and SPN Conversion Method. The DTC together with the source address identifies a component or sub system.

ref:
(https://www.kvaser.com/about-can/higher-layer-protocols/j1939-standards-overview)

DanielMartensson, 2024-01-03:
> J1939/73 Application Layer – Diagnostics This document defines functions and messages for accessing diagnostic and calibration data. There is a number of predefined Diagnostic Messages (DM) used [...]

@payapars 

Yes, but DM14 I have only seen read requests.

payapars, 2024-01-05:
Searching with chat gbt for: how to write memory in J1939 protocol using DM14 DM15 DM16 

[The SAE J1939 protocol includes the DM14, DM15, and DM16 messages for direct memory access](https://wiki.neweagle.net/index.php?title=Raptor-Dev-J1939-Direct-Memory-Access)³. These messages are used to read and write data in the memory of the Electronic Control Unit (ECU)¹³⁴.

Here's a basic overview of how it works:

[1](https://wiki.neweagle.net/index.php?title=Raptor-Dev-J1939-Direct-Memory-Access). **DM14 (Memory Access Request)**: This message is used to request access to a specific memory location in the ECU³⁴. The "pointer" in DM14 represents the address of the memory to be used for the read or write operation¹.

[2](https://electronics.stackexchange.com/questions/516403/what-is-pointer-in-dm14-at-j1939-can-bus-standard). **DM[1](https://wiki.neweagle.net/index.php?title=Raptor-Dev-J1939-Direct-Memory-Access)5 (Memory Access Response)**: This message is sent by the ECU in response to a DM14 message³⁴. It indicates whether the requested memory access operation (read or write) can be performed³⁴.

[3](https://www.dmcinfo.com/services/test-and-measurement-automation/labview-programming/j1939-labview-drivers). **DM[1](https://wiki.neweagle.net/index.php?title=Raptor-Dev-J1939-Direct-Memory-Access)6 (Binary Data Transfer)**: This message is used to transfer the binary data for the read or write operation³⁴. It contains the actual data that is read from or written to the specified memory location³⁴.

Please note that the exact implementation of these messages can vary depending on the specific requirements of your application and the configuration of your ECU³⁴. For more detailed information, you may want to refer to the official SAE J19[3](https://www.dmcinfo.com/services/test-and-measurement-automation/labview-programming/j1939-labview-drivers)9 standards or other resources that provide a comprehensive guide to the J1939 protocol¹²³⁴⁵..

Source: Conversation with Bing, 1/6/[2](https://electronics.stackexchange.com/questions/516403/what-is-pointer-in-dm14-at-j1939-can-bus-standard)024
([1](https://wiki.neweagle.net/index.php?title=Raptor-Dev-J1939-Direct-Memory-Access)) Raptor-Dev-J19[3](https://www.dmcinfo.com/services/test-and-measurement-automation/labview-programming/j1939-labview-drivers)9-Direct-Memory-Access - NewEagleWiki. https://wiki.neweagle.net/index.php?title=Raptor-Dev-J[1](https://wiki.neweagle.net/index.php?title=Raptor-Dev-J1939-Direct-Memory-Access)9[3](https://www.dmcinfo.com/services/test-and-measurement-automation/labview-programming/j1939-labview-drivers)9-Direct-Memory-Access.
(2) protocol - What is "Pointer" in DM[1](https://wiki.neweagle.net/index.php?title=Raptor-Dev-J1939-Direct-Memory-Access)4 at J19[3](https://www.dmcinfo.com/services/test-and-measurement-automation/labview-programming/j1939-labview-drivers)9 CAN-bus standard .... https://electronics.stackexchange.com/questions/5[1](https://wiki.neweagle.net/index.php?title=Raptor-Dev-J1939-Direct-Memory-Access)640[3](https://www.dmcinfo.com/services/test-and-measurement-automation/labview-programming/j1939-labview-drivers)/what-is-pointer-in-dm[1](https://wiki.neweagle.net/index.php?title=Raptor-Dev-J1939-Direct-Memory-Access)4-at-j19[3](https://www.dmcinfo.com/services/test-and-measurement-automation/labview-programming/j1939-labview-drivers)9-can-bus-standard.
(3) XNET J1939 CAN Drivers for LabVIEW | DMC, Inc. - dmcinfo.com. https://www.dmcinfo.com/services/test-and-measurement-automation/labview-programming/j1939-labview-drivers.
(4) NI-CAN Channel J1939 Drivers - dmcinfo.com. https://www.dmcinfo.com/Portals/0/Downloads/Labview/DMC-NI-CAN-Channel-J1939-Drivers-Manual.pdf.
(5) J1939 Diagnostic Messages Extension: Standard - Site. https://www.gridconnect.com/products/j1939-diagnostic-messages-extension.
(6) undefined. http://www.dmcinfo.com.
(7) Getty. https://media.gettyimages.com/photos/unlocked-folder-with-files-in-storage-picture-id1[2](https://electronics.stackexchange.com/questions/516403/what-is-pointer-in-dm14-at-j1939-can-bus-standard)138[4](https://www.dmcinfo.com/Portals/0/Downloads/Labview/DMC-NI-CAN-Channel-J1939-Drivers-Manual.pdf)9430?b=[1](https://wiki.neweagle.net/index.php?title=Raptor-Dev-J1939-Direct-Memory-Access)&k=20&m=121[3](https://www.dmcinfo.com/services/test-and-measurement-automation/labview-programming/j1939-labview-drivers)849430&s=612x612&w=0&h=3_JqPFUeyEGd9Jlenoqt0lyT8BTSvlo2bR3BPA9XM-s=.

DanielMartensson, 2024-01-07:
@payapars Thank you. If you want to implement write operation for DM14, then please, send me a pull request.

payapars, 2024-01-09:
>Thank you. If you want to implement write operation for DM14, then please, send me a pull request.
Yes . Sure

closed 2024-01-10 by DanielMartensson