DanielMartensson/Open-SAE-J1939 #33: different code for Commanded_Address.c

helikl, 2024-01-12:
I've found this file in a link with a bit of different code but the pointer to your repository.
Can you explain what is the difference?
Thank you
Here is the code:
/*
 * SAE_J1939_Commanded_Address.c
 *
 *  Created on: 14 juli 2021
 *      Author: Daniel Mårtensson
 */

#include "Network_Management_Layer.h"

/*
 * Send commanded address to another ECU
 * PGN: 0x00FED8 (65240)
 */
ENUM_J1939_STATUS_CODES SAE_J1939_Send_Commanded_Address(J1939 *j1939, uint8_t DA, uint8_t new_ECU_address, uint32_t identity_number, uint16_t manufacturer_code, uint8_t function_instance, uint8_t ECU_instance, uint8_t function, uint8_t vehicle_system, uint8_t arbitrary_address_capable, uint8_t industry_group, uint8_t vehicle_system_instance) {
	/* Multiple messages - Use Transport Protocol Connection Management BAM */
	uint8_t number_of_packages = 2;
	uint8_t total_message_size = 9;
	uint8_t data[9];
	data[0] = identity_number;
	data[1] = identity_number >> 8;
	data[2] = (identity_number >> 16) |  (manufacturer_code << 5);
	data[3] = manufacturer_code >> 3;
	data[4] = (function_instance << 3) | ECU_instance;
	data[5] = function;
	data[6] = vehicle_system << 1;
	data[7] = (arbitrary_address_capable << 7) | (industry_group << 4) | vehicle_system_instance;
	data[8] = new_ECU_address;								/* New address of the ECU we are sending to*/
	ENUM_J1939_STATUS_CODES status = SAE_J1939_Send_Transport_Protocol_Connection_Management(j1939, DA, CONTROL_BYTE_TP_CM_BAM, total_message_size, number_of_packages, PGN_COMMANDED_ADDRESS);
	if(status != STATUS_SEND_OK)
		return status;
	status = SAE_J1939_Send_Transport_Protocol_Data_Transfer(j1939, DA, data, total_message_size, number_of_packages);
	if(status == STATUS_SEND_OK) {
		/* Send to all ECU that the address DA unused */
		SAE_J1939_Send_Address_Delete(j1939, 0xFF, DA);
		/* According to J1939 standard, request for Address Claimed (broadcast) must be sent after a Commanded Address so we can get back the new ECU address */
		SAE_J1939_Send_Request(j1939, 0xFF, PGN_ADDRESS_CLAIMED);
	}
	return status;
}

/*
 * Read the commanded address from another ECU. Will always be called from Transport Protocol Data Transfer due to 9 bytes of data
 * PGN: 0x00FED8 (65240)
 */
void SAE_J1939_Read_Commanded_Address(J1939 *j1939, uint8_t data[]) {
	j1939->this_name.identity_number = ((data[2] & 0b00011111) << 16) | (data[1] << 8) | data[0];
	j1939->this_name.manufacturer_code = (data[3] << 3) | (data[2] >> 5);
	j1939->this_name.function_instance = data[4] >> 3;
	j1939->this_name.ECU_instance = data[4] & 0b00000111;
	j1939->this_name.function = data[5];
	j1939->this_name.vehicle_system = data[6] >> 1;
	j1939->this_name.arbitrary_address_capable = data[7] >> 7;
	j1939->this_name.industry_group = (data[7] >> 4) & 0b0111;
	j1939->this_name.vehicle_system_instance = data[7] & 0b00001111;
	j1939->this_ECU_address = data[8]; 						/* New address of this ECU */
}

DanielMartensson, 2024-01-13:
What part do you want to have an explanation at?

helikl, 2024-01-13:
BODY { font-family:Arial, Helvetica, sans-serif;font-size:12px; }
 There is this referral to your repositoryThis is how I solved the
issue with a PGN number, made by my self.
github.com/DanielMartensson/Open-SAE-J1939/blob/… [1] – euraad [2]
Aug 11, 2021 at 16:47part of stack overflow discussion.Is adding this
code/* Send to all ECU that the address DA unused
*/SAE_J1939_Send_Address_Delete(j1939, 0xFF, DA); should be the right
way to delte the ECU address?
 Thank you.
 Elik
 On Sat 13/01/24 11:59 PM , Daniel Mårtensson

[4].
[ { ***@***.***": "http://schema.org", ***@***.***": "EmailMessage",
"https://github.com/DanielMartensson/Open-SAE-J1939/issues/33#issuecomment-1890538622",
"url":
"https://github.com/DanielMartensson/Open-SAE-J1939/issues/33#issuecomment-1890538622",
"name": "View Issue" }, "description": "View this Issue on GitHub",
"https://github.com" } } ] 

-------------------------
Msg sent via WebMail

Links:
------
[1]
https://github.com/DanielMartensson/Open-SAE-J1939/blob/3a506fc1d96519a5281991fe401a1500361dbe1a/Src/SAE%20J1939/SAE%20J1939-81%20Network%20Management%20Layer/Commanded_Address.c#L34
[2] https://stackoverflow.com/users/1008405/euraad
[3]
https://github.com/DanielMartensson/Open-SAE-J1939/issues/33#issuecomment-1890538622
[4]

DanielMartensson, 2024-01-14:
> […](#)

@helikl  sorry, i don't understanding your message.

helikl, 2024-01-15:
Sorry, I try to clarify.

There is a discussion at url:

can bus - J1939 - What PGN number should I use for request about delete NAME? - Stack Overflow <https://stackoverflow.com/questions/68337373/j1939-what-pgn-number-should-i-use-for-request-about-delete-name> 

 

As a result the conclusion was to add this line

SAE_J1939_Send_Address_Delete(j1939, 0xFF, DA);

to the file  Commanded Address.c.

In the repository this line does not  show in the file Commanded Address.c..

Is that the right approach for delete the ECU information?

 

Thank you.

 

Elik

 

…

DanielMartensson, 2024-01-15:
@helikl 

"Is that the right approach for delete the ECU information?"

It's not part of the SAE J1939 standard, as explained at the comments. 

The delete ECU PGN is a function call to remove the awareness of the ECU ID from other ECUs.

Example:
If there are 10 ECU with the adresses: 0, 1,..., 9.
And we remove ECU 5. Then we tell the other ECUs that number 5 is gone from the network.

But if ECU 5 is reconnected, then it will broadcast its NAME and all the other ECU will accept this new member.

helikl, 2024-01-16:
Thank you for the explanation and the contribution to use in a simple way the J1939.

Elik

DanielMartensson, 2024-01-17:
Thank you so much. If there are anything else, just send a pull request to contribute to this open library.

closed 2024-01-17 by DanielMartensson