DanielMartensson/Open-SAE-J1939 #34: problem in ENUM_J1939_STATUS_CODES SAE_J1939_Send_Request_DM14 function

payapars, 2024-02-15:
problem in calculate data[1]:

````
ENUM_J1939_STATUS_CODES SAE_J1939_Send_Request_DM14(J1939 *j1939, uint8_t DA, uint16_t number_of_requested_bytes, uint8_t pointer_type, uint8_t command, uint32_t pointer, uint8_t pointer_extension, uint16_t key) {
	uint32_t ID = (0x18D9 << 16) | (DA << 8) | j1939->information_this_ECU.this_ECU_address;
	uint8_t data[8];
	data[0] = number_of_requested_bytes;
	data[1] = (number_of_requested_bytes >> 3) | (pointer_type << 4) | (command << 1) | 0b1;
	data[2] = pointer;
	data[3] = pointer >> 8;
	data[4] = pointer >> 16;
	data[5] = pointer_extension;
	data[6] = key;
	data[7] = key >> 8;
	return CAN_Send_Message(ID, data);
}  ```

if (number_of_requested_bytes >> 3) chage to (number_of_requested_bytes >> 8 )  data is true.

DanielMartensson, 2024-02-15:
If you are sure, please send a pull request

payapars, 2024-02-16:
> If you are sure, please send a pull request

yes I am sure ,but i dont use Github Link.

DanielMartensson, 2024-02-16:
> > If you are sure, please send a pull request
> 
> yes I am sure ,but i dont use Github Link.

All you need to do is to fork my project, change your forked project, then do a pull request. It's easy.

DanielMartensson, 2024-02-16:
@payapars Thank you!

closed 2024-02-16 by DanielMartensson