DanielMartensson/Open-SAE-J1939 #44: Implementation for TI microcontrollers

bharatbohara, 2024-08-09:
Hello,

I am looking how to implement this library on TI microcontroller, specifically, C28x MCUs (F28379D or F28388D). On the repository documentation, I did find its implementation for STM32 microcontrollers and others, however, I could not find any explanation for TI microcontrollers. Could you please help me how can I make it executable for TI MCUs?

Thank you in advance for your support.

Best Regards

DanielMartensson, 2024-08-10:
Just copy the source folder into your project and compile. You cannot use MKL here, only CLAPACK or internal C library of CControl.

bharatbohara, 2024-08-12:
Hi @DanielMartensson,

Thank you very much for your quick reply and the helpful suggestions. I still have some doubts that I hope you can help clarify:

1. I understand that I need to use CLAPACK instead of MKL. Does this require modifying the C and header files within the "Src" folder, or is it sufficient to just link against CLAPACK during the build process?
2. As I don't have much experience with embedded C, could you please provide some guidance on how to proceed with the changes you suggested? Specifically, any steps or best practices for integrating CLAPACK with the existing project would be greatly appreciated.
3. Since the STM32 uses a non-Intel processor, I'm curious how MKL was executable on this platform. Is there something I am missing regarding MKL's compatibility with STM32. Does it uses MKL or CLAPACK instead?

Thank you for your time and assistance. I'm looking forward to your support on how to move forward.

DanielMartensson, 2024-08-12:
@bharatbohara 

1. Yes. Modify the defines.h file at row 46.

2. There are tons of examples inside the example folder.

3. CControl is portable for all systems. Just change your defines.h so CLAPACK or internal is used.

The internal is for the low memory processors.  CLAPACK can be used for 32-bit processors.

DanielMartensson, 2024-08-12:
@bharatbohara 

Oh, sorry. I through you was talking about CControl repository. I was answering this post from my mobile. Did not see the name of the repository.

Yes, you can use Open SAE J1939 on all systems. Just copy the source folder and see the examples.

Begin with the hardware.h and change it to INTERNAL CALLBACK.

Then you loading the example "internal callback.txt".

bharatbohara, 2024-08-12:
@DanielMartensson 

Thank you for your quickest reply I have ever got.

Sure, I will try this example on my TI microcontroller.

Thank you. I appreciate your help.

bharatbohara, 2024-08-12:
@DanielMartensson 

Hi Daniel,

Now, I am able to build and load the library on the MCU, however, I am not able to send any messages yet. For instance, I want to generate a sinusoidal signal on MCU and pass it through CAN using J1939 protocol and observe the waveform on MATLAB Simulink using this example - https://www.mathworks.com/help/vnt/ug/basic-j1939-communication-over-can.html (only the receiver part on PC, while sender would be on MCU). Another problem I am trying to do is getting raw data from ECU, for that how can I change the ID of ECU on Internal Callback example file, please? Is there any other examples which I should look through to fully understand the functionality of this library?

Note: I want the MCU to both send and receive the message using J1939 protocol through CAN bus.

Thank you for your support.

DanielMartensson, 2024-08-12:
@bharatbohara 

If you want to send arbitrary raw data, then use the DM16 example.

bharatbohara, 2024-08-12:
@DanielMartensson,

Sorry I mean to say just the signal data for the sinusoidal wave generated on MCU. Here is the snippet of the code with sine wave and I am trying to send with PGN #0x009D00U.

` int main() {
 
/* Create our J1939 structure with two ECU */
    J1939 j1939_1 = { 0 };

    /* Important to sent all non-address to 0xFF - Else we cannot use ECU address 0x0 */
    uint8_t i;
    for (i = 0; i < 255; i++) {
        j1939_1.other_ECU_address[i] = 0xFF;
    }

    /* Set the ECU address */
    j1939_1.information_this_ECU.this_ECU_address = 0x80; /* From 0 to 253 because 254 = error address and 255 = broadcast address */

    /* Set the callback functions for the hardware */
    CAN_Set_Callback_Functions(Callback_Function_Send, Callback_Function_Read, NULL, Callback_Function_Delay);

    /* Set NAME for ECU 2 by using Commanded Address. Also change the ECU address from 0x90 to 0x91 */
    SAE_J1939_Send_Commanded_Address(&j1939_1, 0x90, 0x91, 5000, 500, 30, 5, FUNCTION_AUXILIARY_VALVES_CONTROL, 50, 0, INDUSTRY_GROUP_AGRICULTURAL_AND_FORESTRY, 15);

    /* Send a PGN request */
    SAE_J1939_Send_Request(&j1939_1, 0x20, PGN_MATLAB_SEND);

    /* Read a SAE J1939 message */
    /*Open_SAE_J1939_Listen_For_Messages(&j1939_1);*/

    /* Main loop */
        float time = 0;
        while (1) {
            /* Generate sinusoidal signal value */
            float signal_value = Generate_Sinusoidal_Signal(time);

            /* Pack the signal value into an 8-byte J1939 message */
            uint8_t data[8] = { 0 };
            uint16_t packed_value = (uint16_t)(signal_value + AMPLITUDE); // Offset to ensure positive values
            data[0] = (packed_value >> 8) & 0xFF; // High byte
            data[1] = packed_value & 0xFF;        // Low byte

            /* Send the CAN message using J1939 protocol */
            SAE_J1939_Send_Request(&j1939_1, j1939_1.information_this_ECU.this_ECU_address, packed_value);

            /* Increment time for the next sample */
            time += 1.0 / SAMPLING_RATE;

            /* Apply a delay to match the sampling rate */
            Callback_Function_Delay(1);
        }

    return 0;
} `

Here are the parameters for J1939 interface in Matlab with PGN #40192 and node address of 0x80. I tried to configure the example code for internal callback.c with these parameters. Is there anything I am missing?
[image]
[image]

Any kind of suggestions and support is greatly appreciated.

Thank you.

DanielMartensson, 2024-08-12:
Here is a better one

```c
 int main() {

/* Create our J1939 structure with two ECU */
J1939 j1939_1 = { 0 };

/* Important to sent all non-address to 0xFF - Else we cannot use ECU address 0x0 */
uint8_t i;
for (i = 0; i < 255; i++) {
    j1939_1.other_ECU_address[i] = 0xFF;
}

/* Set the ECU address - You can use all numbers between 0 and 253! */
j1939_1.information_this_ECU.this_ECU_address = 0x80; /* From 0 to 253 because 254 = error address and 255 = broadcast address */

/* Set the callback functions for the hardware - Good! */
CAN_Set_Callback_Functions(Callback_Function_Send, Callback_Function_Read, NULL, Callback_Function_Delay);

/* Set NAME for ECU 2 by using Commanded Address. Also change the ECU address from 0x90 to 0x91 - This is only used if you want to change another ECU's NAME...NAME is the ID of the device e.g what pourpose it should have and where it exist. Read the Open SAE J1939.pdf file and you will understand what values you can use. Or read the header C files. */
SAE_J1939_Send_Commanded_Address(&j1939_1, 0x90, 0x91, 5000, 500, 30, 5, FUNCTION_AUXILIARY_VALVES_CONTROL, 50, 0, INDUSTRY_GROUP_AGRICULTURAL_AND_FORESTRY, 15);

/* Main loop */
    float time = 0;
    while (1) {
        /* Read a SAE J1939 message from another ECU, if needed.  */
        /*Open_SAE_J1939_Listen_For_Messages(&j1939_1);*/
        
        /* Generate sinusoidal signal value */
        float signal_value = Generate_Sinusoidal_Signal(time);

        /* Pack the signal value into an 8-byte J1939 message */
        uint8_t data[2] = { 0 };
        uint8_t number_of_occurences = 2;
        uint16_t packed_value = (uint16_t)(signal_value + AMPLITUDE); // Offset to ensure positive values
        data[0] = (packed_value >> 8) & 0xFF; // High byte
        data[1] = packed_value & 0xFF;        // Low byte

        /* Send the CAN message using J1939 protocol */
        uint8_t DA = address of the other ECU that going to have the message
        SAE_J1939_Send_Binary_Data_Transfer_DM16(&j1939_1, DA, number_of_occurences, data);

        /* Increment time for the next sample */
        time += 1.0 / SAMPLING_RATE;

        /* Apply a delay to match the sampling rate */
        Callback_Function_Delay(1);
    }

return 0;
} 

```

bharatbohara, 2024-08-13:
@DanielMartensson 

Thank you very much for updating my code. However, I am getting some error on my MCU IDE, and I am trying to find some help from Texas Instruments forum.

> ERROR: system post cinit() an error occured failed to parse the previous frame FP

In my opinion, this might be because the "Callback_Function_Delay" is not returning anything. I am trying to figure out how to add delay function on TI boards. I have tried with `DELAY_US(delay)`, but its throwing me something else error. I will try to debug this error and let you know if the above code works.

Thank you again for your kind help. I appreciate your support.

DanielMartensson, 2024-08-13:
It's not a library issue. Post your whole code here.

bharatbohara, 2024-08-13:
Here is my whole code:

```/*
 * Main.c
 *
 *  Created on: 16 July 2021
 *      Author: Daniel Mårtensson
 */

#include <stdio.h>
#include <math.h>
#include "device.h"  // Include this for the DELAY_US function on TI microcontrollers

/* Include Open SAE J1939 */
#include "Open_SAE_J1939/Open_SAE_J1939.h"

#define PI 3.14159265
#define SAMPLING_RATE 1000 // Samples per second
#define AMPLITUDE 100      // Max amplitude of the sine wave
#define FREQUENCY 1        // Frequency of the sine wave in Hz

/* Every time we want to send a CAN message, this function will be called */
void Callback_Function_Send(uint32_t ID, uint8_t DLC, uint8_t data[]) {
    printf("Callback_Function_Send called\n");
    printf("ID = 0x%X\n", ID);
    printf("DLC = %i\n", DLC);
    uint8_t i;
    for (i = 0; i < DLC; i++) {
        printf("data[%i] = 0x%X\n", i, data[i]);
    }
    printf("\n");
}

/* Every time we want to read a CAN message, this function will be called */
void Callback_Function_Read(uint32_t* ID, uint8_t data[], bool* is_new_data) {
    printf("Callback_Function_Read called\n");
    *ID = 0xFF;
    uint8_t i;
    for (i = 0; i < 8; i++) {
        data[i] = 0xFF;
    }
    *is_new_data = true;
}

/* Apply your delay here */
void Callback_Function_Delay(uint8_t delay){
    // Convert the delay to microseconds (assuming delay is in milliseconds)
    uint32_t delay_in_us = (uint32_t)delay * 1000;

    // Use the DELAY_US function provided by TI libraries
    DELAY_US(delay_in_us);
}

/* Function to generate and return a sinusoidal value */
float Generate_Sinusoidal_Signal(float time) {
    return AMPLITUDE * sin(2 * PI * FREQUENCY * time);
}

int main() {
    /* Create our J1939 structure with two ECU */
    J1939 j1939_1 = {0};

    /* !!!!! DON'T FORGET TO CHANGE THE PROCESSOR_CHOICE to INTERNAL_CALLBACK in Hardware.h !!!!! */

    /* Important to set all non-addresses to 0xFF - Else we cannot use ECU address 0x0 */
    uint8_t i;
    for (i = 0; i < 255; i++) {
        j1939_1.other_ECU_address[i] = 0xFF;
    }

    /* Set the ECU address - You can use all numbers between 0 and 253 */
    j1939_1.information_this_ECU.this_ECU_address = 0x00; /* From 0 to 253 because 254 = error address and 255 = broadcast address */

    /* Set the callback functions for the hardware */
    CAN_Set_Callback_Functions(Callback_Function_Send, Callback_Function_Read, NULL, Callback_Function_Delay);

    /* Main loop */
    float time = 0.0;
    while (1) {
        /* Generate sinusoidal signal value */
        float signal_value = Generate_Sinusoidal_Signal(time);

        /* Pack the signal value into an 8-byte J1939 message */
        uint8_t data[2] = {0};
        uint8_t number_of_occurrences = 2;
        uint16_t packed_value = (uint16_t)(signal_value + AMPLITUDE); // Offset to ensure positive values
        data[0] = (packed_value >> 8) & 0xFF; // High byte
        data[1] = packed_value & 0xFF;        // Low byte

        /* Send the CAN message using J1939 protocol */
        uint8_t DA = 0x50; // address of the other ECU that is going to receive the message
        SAE_J1939_Send_Binary_Data_Transfer_DM16(&j1939_1, DA, number_of_occurrences, data);

        /* Increment time for the next sample */
        time += 1.0 / SAMPLING_RATE;

        /* Apply a delay to match the sampling rate */
        Callback_Function_Delay(1);
    }

    return 0; 
}```

DanielMartensson, 2024-08-13:
It's just a compiler issue. Not wrong with your code.

bharatbohara, 2024-08-13:
@DanielMartensson , Okay sure please. I will remove floats and try to recompile it. Thank you for your feedback.

bharatbohara, 2024-08-13:
@DanielMartensson,

I tried with increasing the stack memory. But still I am getting the same error. One strange thing that I have experience is: if I include `#include "driverlib.h", #include "device.h"` in order to use the inbuilt delay function, I get some datatype redeclaration error. On the other hand, if I comment them, I get the above error.

For you reference, if I am trying to include the above device specific libraries, I get the following errors. Have you encounter any problem like this before?

``` 
>> Compilation failure
subdir_rules.mk:9: recipe for target 'Internal_callback.obj' failed
"..\Open_SAE_J1939\C89_Library.h", line 25: error #258: invalid redeclaration of type name "uint8_t" (declared at line 49 of "C:\ti\c2000\C2000Ware_4_03_00_00\driverlib\f2838x\driverlib\inc/hw_types.h")
"..\Open_SAE_J1939\C89_Library.h", line 26: error #258: invalid redeclaration of type name "uint16_t" (declared at line 65 of "C:/ti/ccs1220/ccs/tools/compiler/ti-cgt-c2000_22.6.0.LTS/include/sys/_stdint.h")
"..\Open_SAE_J1939\C89_Library.h", line 27: error #258: invalid redeclaration of type name "uint32_t" (declared at line 70 of "C:/ti/ccs1220/ccs/tools/compiler/ti-cgt-c2000_22.6.0.LTS/include/sys/_stdint.h")
"..\Open_SAE_J1939\C89_Library.h", line 28: error #258: invalid redeclaration of type name "int8_t" (declared at line 50 of "C:\ti\c2000\C2000Ware_4_03_00_00\driverlib\f2838x\driverlib\inc/hw_types.h")
"..\Open_SAE_J1939\C89_Library.h", line 29: error #258: invalid redeclaration of type name "int16_t" (declared at line 43 of "C:/ti/ccs1220/ccs/tools/compiler/ti-cgt-c2000_22.6.0.LTS/include/sys/_stdint.h")
"..\Open_SAE_J1939\C89_Library.h", line 30: error #258: invalid redeclaration of type name "int32_t" (declared at line 48 of "C:/ti/ccs1220/ccs/tools/compiler/ti-cgt-c2000_22.6.0.LTS/include/sys/_stdint.h")
"..\Open_SAE_J1939\C89_Library.h", line 33: error #85: invalid combination of type specifiers
"../Internal_callback.c", line 236: warning #112-D: statement is unreachable
7 errors detected in the compilation of "../Internal_callback.c".
gmake: *** [Internal_callback.obj] Error 1
gmake: Target 'all' not remade because of errors.

**** Build Finished ****

DanielMartensson, 2024-08-13:
Your compiler cannot find the header files. Notice that even if your IDE can find the header files, it does not mean that the compiler have a correct setup of the header files.

bharatbohara, 2024-08-13:
Sorry I could not find the header files folder, neither defines.h

DanielMartensson, 2024-08-14:
Try structs.h file? There J1939 should be.

Do you have UART communication? If yes, then you can output the traffic by using a calback function instead of NULL that you are using now.

The library is simple. It's just a layer. You are connecting the layer to your hardware CAN functions by using calllback functions.

Then you just using the examples that can send requests or transfer data. Super easy @bharatbohara

DanielMartensson, 2024-08-14:
@Varat7v2 
Just fork the project.
Change the code inside your forked project.
Then go to pull request and send your forked project to my project.

I will then get a notification.

Varat7v2, 2024-08-14:
@DanielMartensson Did you receive the notification for pull request, please?

DanielMartensson, 2024-08-14:
> @DanielMartensson Did you receive the notification for pull request, please?

@Varat7v2 yes. I will have a look at it

closed 2024-08-25 by DanielMartensson