DanielMartensson/Open-SAE-J1939 #49: Is TP.CM_RTS implemented correctly?

alexch-m, 2024-10-28:
Hello, Daniel!

I played with DM1 example to validate my own partial implementation of J1939 and found that possibly TP.CM works not according to the standard (sorry if I'm wrong - I'm a beginning in J1939). The test app is a modified version of your DM1 example which uses Socket CAN (vcan) and supports two modes (request and reply) for easy debugging. This is a log of CAN frames sent in and out when one instance of the test app requests DM1 from another one:
```
-> ID = 0x18EA5A6B, DLC = 3 [CA FE 00]
<- ID = 0x1CEC6B5A, DLC = 8, data = [10 12 00 03 FF CA FE 00]
-> ID = 0x1CEC5A6B, DLC = 8 [11 00 00 FF FF CA FE 00]
<- ID = 0x1CEB6B5A, DLC = 8, data = [01 44 11 6C 02 06 01 37]
-> ID = 0x1CEC5A6B, DLC = 8 [11 01 01 FF FF CA FE 00]
<- ID = 0x1CEB6B5A, DLC = 8, data = [02 04 04 02 6F 00 10 83]
-> ID = 0x1CEC5A6B, DLC = 8 [11 02 02 FF FF CA FE 00]
<- ID = 0x1CEB6B5A, DLC = 8, data = [03 46 00 0B 82 FF FF FF]
-> ID = 0x1CEC5A6B, DLC = 8 [13 12 00 03 FF CA FE 00]
```
Looking at TP.CM_CTS frames (starting from 0x11), we see that bytes n.2 and n.3 (numeration starts from 1) are "00 00", "01 01", "02 02". According to SAE J1939-21, byte n.2 is a "_Number of packets that can be sent. This value shall be no larger than the value in byte 5 of the RTS message._" I'd expect to see e.g. value 1 here (if the party which sends CTS wants to receive frames one-by-one), but we see numbers identical to byte n.3 ("_Next packet number to be sent_").

The code in Transport_Protocol_Connection_Management.c:
```
	case CONTROL_BYTE_TP_CM_CTS:
		data[1] = j1939->this_ecu_tp_cm.total_number_of_packages_transmitted;
		data[2] = j1939->this_ecu_tp_cm.next_packet_number_transmitted;

```
-- is it correct?

DanielMartensson, 2024-10-28:
Hi!

@alexch-m 

- Byte 5 is reserved
- Byte 2 and 3 are number of bytes
- Byte 4 is number of packages

Se here:
https://github.com/DanielMartensson/Open-SAE-J1939/blob/main/Src/Documentation/Pictures/SAE%20J1939%20Resources/TCI%20%20SENSE%2042%20Version%203.0-25-108.pdf

Notice! That's for BAM messages, but I'm expecting BAM and CTS to have the same data frame.

alexch-m, 2024-10-29:
Hi! According to info I've found so far that's different, please look at the picture below (TP.CM_CTS description). The picture is from [SAE-J1939-21-2006.pdf](https://tajhizkala.ir/doc/SAE/SAE-J1939-21-2006.pdf). And other libraries I tried (e.g. [python-can-j1939](https://github.com/juergenH87/python-can-j1939)) work differently to your implementation. I can provide CAN dumps later if needed. Unfortunately I don't have examples of TP.CM communications from real hardware yet.
[image]

DanielMartensson, 2024-10-29:
@alexch-m 

The reason why you might be confused, is because I'm re-using the same data frame and only change some new numbers. 
As you can see, byte 6 to byte 8 (indexing from 1) are the same for BAM, Conn_Abort, EndOfMsgACK, CTS and RTS.

https://github.com/DanielMartensson/Open-SAE-J1939/blob/9ab0d03f9eb7bc5dac09e8476d9239ce4e504825/Src/SAE_J1939/SAE_J1939-21_Transport_Layer/Transport_Protocol_Connection_Management.c#L26-L52

RamDav, 2025-05-08:
I have the same issue i'm using a python implementation of a j1939 stack on a windows machine and logging the communication via PCAN Viewer. I have this stack ported to my mcu which runs on a Microchip SAM  processor. 
This is the python stack:
https://github.com/juergenH87/python-can-j1939/tree/master
I'm trying to read the embedded ecus software identification via python script.
I've set my Software id of this ecu to something larger than 2 * 7 bytes to have a test for multipacket transfers.

* Request
Upon Requesting the Software ID via python script this message is send by the pc:
ID 0x18EAA14F with data [DA, FE, 00] 
which is the correct request for  PGN "FEDA".
 
* Request to Send
This is then followed by the response from this stack 
ID 0x1CEC4FA1 data [0x10, 0x15, 0x00, 0x03, 0xFF, 0xDA, 0xFE, 0x00]
which corresponds to request to send 0x15 bytes of data in 3 packages and 0xFF packages max transfer at once for pgn 0xFEDA
 
* Clear to Send 1 package starting with 1
which is followed by the clear to send from the python stack
ID 0x1CECA14F data [0x11, 0x01, 0x01, 0xFF, 0xFF, 0xDA, 0xFE, 0x00]
clear to send 1 package next package is number 1  for the pgn
 
* Data transfer 0
this lead to this stack sending the second part of the package 
ID 0x1CEB4FA1 data [0x02, <7 Bytes of data>]
which is the package number 2 (shouldn't this be number 1) ? 

* Clear to Send 1 package starting with 2
the pyton stack continues with cts 
ID 0x1CECA14F data [0x11, 0x1, 0x02 ,0xFF, 0xFF, 0xDA, 0xFE, 0x00]
clear to send one package next number should be two

* Data Transfer 1
which is answered by this stack with 
ID 0x1CEB4FA1 data [0x03, <7 Bytes of data>]
This should be package 2 not 3.

This is caused by this stack interpreting byte 1 of the cts data as number of already send packages  which leads to it starting at the wrong and always the same offset in the data to be send.

reading the byte from the cts message:
```c
void SAE_J1939_Read_Transport_Protocol_Connection_Management(J1939 *j1939, uint8_t SA, uint8_t data[]) {
....
case CONTROL_BYTE_TP_CM_CTS:
		j1939->from_other_ecu_tp_cm.total_number_of_packages_transmitted = data[1]; // this is the number of packages which should be send not already send
		j1939->from_other_ecu_tp_cm.next_packet_number_transmitted = data[2];
		SAE_J1939_Send_Transport_Protocol_Data_Transfer(j1939, SA);
```
```c
ENUM_J1939_STATUS_CODES SAE_J1939_Send_Transport_Protocol_Data_Transfer(J1939 *j1939, uint8_t DA){
.....
case CONTROL_BYTE_TP_CM_CTS:
		package[0] = j1939->from_other_ecu_tp_cm.next_packet_number_transmitted + 1;				/* Next number of package */ // !!!!this should not be increased by one
		bytes_sent = j1939->from_other_ecu_tp_cm.total_number_of_packages_transmitted * 7;  // !!!this really is the number of packages need to be send the number of bytes send is next packet number * 7  this lead to always sending the second set of seven bytes.
		for (j = 0; j < 7; j++) {
			if (bytes_sent < j1939->this_ecu_tp_cm.total_message_size_being_transmitted) {
				package[j + 1] = j1939->this_ecu_tp_dt.data[bytes_sent++];						/* Data that we have collected */
			}
			else {
				package[j + 1] = 0xFF; 															/* Reserved */
			}
		}

		/* Transmitt message */
		status = CAN_Send_Message(ID, package);
		break;
	}

```

Your table in your document differs from the table from the official SAE-J1939 definition as shown by the original poster.
And furthermore this stack sets the number of Max Packages to MAX which is not handled by this stack as is just can transfer one package per CTS Request.

DanielMartensson, 2025-05-08:
> I have the same issue i'm using a python implementation of a j1939 stack on a windows machine and logging the communication via PCAN Viewer. I have this stack ported to my mcu which runs on a Microchip SAM processor. This is the python stack: https://github.com/juergenH87/python-can-j1939/tree/master I'm trying to read the embedded ecus software identification via python script. I've set my Software id of this ecu to something larger than 2 * 7 bytes to have a test for multipacket transfers.
> 
> * Request
>   Upon Requesting the Software ID via python script this message is send by the pc:
>   ID 0x18EAA14F with data [DA, FE, 00]
>   which is the correct request for  PGN "FEDA".
> * Request to Send
>   This is then followed by the response from this stack
>   ID 0x1CEC4FA1 data [0x10, 0x15, 0x00, 0x03, 0xFF, 0xDA, 0xFE, 0x00]
>   which corresponds to request to send 0x15 bytes of data in 3 packages and 0xFF packages max transfer at once for pgn 0xFEDA
> * Clear to Send 1 package starting with 1
>   which is followed by the clear to send from the python stack
>   ID 0x1CECA14F data [0x11, 0x01, 0x01, 0xFF, 0xFF, 0xDA, 0xFE, 0x00]
>   clear to send 1 package next package is number 1  for the pgn
> * Data transfer 0
>   this lead to this stack sending the second part of the package
>   ID 0x1CEB4FA1 data [0x02, <7 Bytes of data>]
>   which is the package number 2 (shouldn't this be number 1) ?
> * Clear to Send 1 package starting with 2
>   the pyton stack continues with cts
>   ID 0x1CECA14F data [0x11, 0x1, 0x02 ,0xFF, 0xFF, 0xDA, 0xFE, 0x00]
>   clear to send one package next number should be two
> * Data Transfer 1
>   which is answered by this stack with
>   ID 0x1CEB4FA1 data [0x03, <7 Bytes of data>]
>   This should be package 2 not 3.
> 
> This is caused by this stack interpreting byte 1 of the cts data as number of already send packages which leads to it starting at the wrong and always the same offset in the data to be send.
> 
> reading the byte from the cts message:
> 
> void SAE_J1939_Read_Transport_Protocol_Connection_Management(J1939 *j1939, uint8_t SA, uint8_t data[]) {
> ....
> case CONTROL_BYTE_TP_CM_CTS:
> 		j1939->from_other_ecu_tp_cm.total_number_of_packages_transmitted = data[1]; // this is the number of packages which should be send not already send
> 		j1939->from_other_ecu_tp_cm.next_packet_number_transmitted = data[2];
> 		SAE_J1939_Send_Transport_Protocol_Data_Transfer(j1939, SA);
> ENUM_J1939_STATUS_CODES SAE_J1939_Send_Transport_Protocol_Data_Transfer(J1939 *j1939, uint8_t DA){
> .....
> case CONTROL_BYTE_TP_CM_CTS:
> 		package[0] = j1939->from_other_ecu_tp_cm.next_packet_number_transmitted + 1;				/* Next number of package */ // !!!!this should not be increased by one
> 		bytes_sent = j1939->from_other_ecu_tp_cm.total_number_of_packages_transmitted * 7;  // !!!this really is the number of packages need to be send the number of bytes send is next packet number * 7  this lead to always sending the second set of seven bytes.
> 		for (j = 0; j < 7; j++) {
> 			if (bytes_sent < j1939->this_ecu_tp_cm.total_message_size_being_transmitted) {
> 				package[j + 1] = j1939->this_ecu_tp_dt.data[bytes_sent++];						/* Data that we have collected */
> 			}
> 			else {
> 				package[j + 1] = 0xFF; 															/* Reserved */
> 			}
> 		}
> 
> 		/* Transmitt message */
> 		status = CAN_Send_Message(ID, package);
> 		break;
> 	}
> Your table in your document differs from the table from the official SAE-J1939 definition as shown by the original poster. And furthermore this stack sets the number of Max Packages to MAX which is not handled by this stack as is just can transfer one package per CTS Request.

@RamDav 

Tänk you. Can you do a PR for this?

DRam86, 2025-05-09:
added pull request #67

DanielMartensson, 2025-05-12:
Pull request accepted

closed 2025-05-12 by DanielMartensson