DanielMartensson/Open-SAE-J1939 #56: Elements of Component_Identification(PGN 0xFEEB)

ChunBoo, 2024-12-16:
Hi Daniel,
    Thank you for sharing this project.
    One issue found in the struct [Component_identification ](https://github.com/DanielMartensson/Open-SAE-J1939/blob/main/Src/Open_SAE_J1939/Structs.h#L173), the first element seems is Make(SPN 586), but not array for "product date", not sure this one induced by different version?

DanielMartensson, 2024-12-16:
@ChunBoo 
Can you provide me some examples or info where its stand Make(SPN)?

ChunBoo, 2024-12-16:
> @ChunBoo Can you provide me some examples or info where its stand Make(SPN)?

Hi @Daniel, please see below screenshot:
[image]

ChunBoo, 2024-12-16:
And below is codes snippet from gpt
typedef struct {
    // Extended Information
    char manufacturer_name[32];
    char component_model[32];
    char serial_number[32];
    char unit_name[32];
    uint8_t certification_data[16];
} J1939_Extended_ID;

DanielMartensson, 2024-12-16:
> > @ChunBoo Can you provide me some examples or info where its stand Make(SPN)?
> 
> Hi @DanielMartensson , please see below screenshot:
> [image]
> 

@ChunBoo 
That must be product date.

A lot of companies have their own modification of the standard.

ChunBoo, 2024-12-16:
> @ChunBoo That must be product date.
>
> A lot of companies have their own modification of the standard.

Thanks, Daniel. The standard of J1939 define this product date in the Component Identification, some companies modified it as manufacturer. Got it now.

closed 2024-12-16 by ChunBoo