DanielMartensson/Open-SAE-J1939 #58: Value of message ID  in Open SAE J1939

ChunBoo, 2024-12-19:
Hi Daniel,
   May I know the value of 0x18EA in below function is our own defined? If yes, can we change it as other value?
ENUM_J1939_STATUS_CODES SAE_J1939_Send_Request(J1939 *j1939, uint8_t DA, uint32_t PGN_code) {
	uint8_t PGN[3];
	PGN[0] = PGN_code;														/* PGN least significant bit */
	PGN[1] = PGN_code >> 8;													/* PGN mid bit */
	PGN[2] = PGN_code >> 16;												/* PGN most significant bit */
	uint32_t ID = (**0x18EA** << 16) | (DA << 8) | j1939->information_this_ECU.this_ECU_address;
	return CAN_Send_Request(ID, PGN);
}

Thanks,
Bob

DanielMartensson, 2024-12-19:
This ID code cannot be changed. It's a part of the standard. @ChunBoo

ChunBoo, 2024-12-20:
> This ID code cannot be changed. It's a part of the standard. @ChunBoo

@DanielMartensson Thanks,Daniel. Merry Christmas！

closed 2024-12-20 by ChunBoo