DanielMartensson/Open-SAE-J1939 #59: cts

TracyCurry, 2024-12-20:
I find a little differenct definition between "j1939-2006-21.pdf" and your "Open SAE J1939.pdf". In "j1939-2006-21.pdf", it says byte2 of cts is "Number of packets that can be sent", and i think it should means that receiver tells the sender that the max number of bytes it can receive instead of "total packages we have received" mentioned in "Open SAE J1939.pdf"
[image]
[image]
pic1 is from "j1939-2006-21.pdf" and pic2 is from "Open SAE J1939.pdf"

TracyCurry, 2024-12-20:
[image]
and in the code, you also set byte2 of cts for the first time as 0 which is exactly the meaning of "total packages we have received". that make me very confused

DanielMartensson, 2024-12-20:
When, when total packages transmitted is equal as the total message, then the transition is complete.

closed 2024-12-20 by TracyCurry