DanielMartensson/Open-SAE-J1939 #60: Is the number_of_packages_being_transmitted should be total_message_size_being_transmitted/7?

ChunBoo, 2025-01-03:
Hi Daniel,

   In [this](https://github.com/DanielMartensson/Open-SAE-J1939/blob/main/Src/SAE_J1939/SAE_J1939-71_Application_Layer/Request_Component_Identification.c#L55) line, it seems the number_of_packages_being_transmitted should be total_message_size_being_transmitted/7, but not total_message_size_being_transmitted/8? 
   As the first byte of each package will be used as the sequence number, please correct me if I am wrong.Thanks.

DanielMartensson, 2025-01-03:
@ChunBoo 
I did this because you need always and extra package if you're sending more than 8 bytes

ChunBoo, 2025-01-04:
@DanielMartensson  seems [this](https://github.com/DanielMartensson/Open-SAE-J1939/blob/main/Src/SAE_J1939/SAE_J1939-73_Diagnostics_Layer/DM2.c#L42) line uses different value(7,but not 8).

DanielMartensson, 2025-01-05:
@ChunBoo 
Can you try to print some CAN traffic to see if it's 8 bytes per package?

I remember that I got a PR to fix the DM code once. Perhaps it was changed to 7 from 8 for some reason.

closed 2025-01-06 by ChunBoo