DanielMartensson/Open-SAE-J1939 #61: Is this one bug when sending BAM message?

ChunBoo, 2025-01-10:
Hi Daniel,
   For the case of transmitting BAM message ,in [this](https://github.com/DanielMartensson/Open-SAE-J1939/blob/main/Src/SAE_J1939/SAE_J1939-21_Transport_Layer/Transport_Protocol_Data_Transfer.c#L111), it seems the sequence number will be always 1, then the receiver can not be terminated, since [this condition](https://github.com/DanielMartensson/Open-SAE-J1939/blob/main/Src/SAE_J1939/SAE_J1939-21_Transport_Layer/Transport_Protocol_Data_Transfer.c#L28) can not be satisfied.
  Not sure this is one issue or not. Thanks.

Best Regards,
Bob

ChunBoo, 2025-01-10:
Seems this case can not be happened most of time, so closed it.

DanielMartensson, 2025-01-10:
@ChunBoo show me your example code.

DanielMartensson, 2025-01-11:
> > @ChunBoo show me your example code.
> > Hi, Daniel, just from the source codes execution sequences, I am not very sure this case will be happened or not(will implement some codes to verify if  such a scenario exists). Suppose one device reports two errors at the same time which need to broadcast to other devices, then those two active errors will be sent as two packets, then above issue will be found.

@ChunBoo 
I cannot see your source code.

ChunBoo, 2025-01-13:
> @ChunBoo I cannot see your source code.
Please see below codes:
       //fill two FMIs for j1939_1 
	j1939_1.this_dm.dm1.FMI[0] = FMI_NOT_AVAILABLE; /* If FMI_NOT_AVAILABLE, then errors_dm1_active will become 0 */
	j1939_1.this_dm.dm1.SPN[0] = 0xA7;
	j1939_1.this_dm.dm1.SPN_conversion_method[0] = 1;
	j1939_1.this_dm.dm1.occurrence_count[0] = 1;

	j1939_1.this_dm.dm1.FMI[1] = 0X10; /* If FMI_NOT_AVAILABLE, then errors_dm1_active will become 0 */
	j1939_1.this_dm.dm1.SPN[1] = 0X6E;//SPN_SENSOR_SUPPLY_VOLTAGE_1;
	j1939_1.this_dm.dm1.SPN_conversion_method[1] = 1;
	j1939_1.this_dm.dm1.occurrence_count[1] = 2;
        j1939_1.this_dm.errors_dm1_active = 2

     //send the message as BAM
     SAE_J1939_Send_Request(j1939_1, **0xFF**, PGN_DM1)

DanielMartensson, 2025-01-13:
@ChunBoo 

Can you fix the code message? It seems that you have not used the code tag properly.

ChunBoo, 2025-01-13:
> @ChunBoo
> 
> Can you fix the code message? It seems that you have not used the code tag properly.

@DanielMartensson  Hi, Daniel, could you please share more details for how to use the code tag?  Not very clear about "fix the code message" here. Thanks.

DanielMartensson, 2025-01-13:
@ChunBoo 

Use the ```

ChunBoo, 2025-01-13:
> @ChunBoo
> 
> Use the ```
@DanielMartensson , thanks! Updated as below lines with  ``` :)
 ```
       //fill two FMIs for j1939_1 
	j1939_1.this_dm.dm1.FMI[0] = 0x11; /* If FMI_NOT_AVAILABLE, then errors_dm1_active will become 0 */
	j1939_1.this_dm.dm1.SPN[0] = 0xA7;
	j1939_1.this_dm.dm1.SPN_conversion_method[0] = 1;
	j1939_1.this_dm.dm1.occurrence_count[0] = 1;

	j1939_1.this_dm.dm1.FMI[1] = 0X10; /* If FMI_NOT_AVAILABLE, then errors_dm1_active will become 0 */
	j1939_1.this_dm.dm1.SPN[1] = 0X6E;//SPN_SENSOR_SUPPLY_VOLTAGE_1;
	j1939_1.this_dm.dm1.SPN_conversion_method[1] = 1;
	j1939_1.this_dm.dm1.occurrence_count[1] = 2;
        j1939_1.this_dm.errors_dm1_active = 2

     //send the dm1 message as BAM
     SAE_J1939_Send_Request(j1939_1, 0xFF, PGN_DM1)
 ```

DanielMartensson, 2025-01-13:
@ChunBoo 

I don't see any issues with that code. Everything is working.

closed 2025-01-14 by ChunBoo