DanielMartensson/Open-SAE-J1939 #62: Is there any suggested time to update dm2 value?

ChunBoo, 2025-01-15:
Hi Daniel,

    For the time of updating current ecu's dm2, can we copy the new active dm1 values to dm2 directly at the end of updating dm1? likes below lines:
   ```
	j1939_1.this_dm.dm2.FMI[0]=j1939_1.this_dm.dm1.FMI[0] = FMI_NOT_AVAILABLE; 
	j1939_1.this_dm.dm2.SPN[0]=j1939_1.this_dm.dm1.SPN[0] = 0xA7;
	j1939_1.this_dm.dm2.SPN_conversion_method[0] =j1939_1.this_dm.dm1.SPN_conversion_method[0] = 1;
	j1939_1.this_dm.dm2.occurrence_count[0]=j1939_1.this_dm.dm1.occurrence_count[0] = 1;       
```
   Or we can update dm2 after responding the request of  dm1 ?   Thanks in advance.

Best Regards,
Bob

DanielMartensson, 2025-01-15:
@ChunBoo 
Yes. You can do that inside your micro controller.

ChunBoo, 2025-01-15:
> @ChunBoo Yes. You can do that inside your micro controller.

Thanks ,Daniel.

closed 2025-01-15 by ChunBoo