DanielMartensson/Open-SAE-J1939 #63: acknowledgement destination address

TracyCurry, 2025-02-06:
In j1939-21 specificaion, it says that for acknowledgement PGN 59392, it's destination address should be a global address which is 255 and in this code, I see that it use the address of ecu which requests an acknowledgement as  destination address when reply back PGN 59392, is it that ok with j1939 specification or there is something wrong about my understanding of j1939 or the code?

[image]

DanielMartensson, 2025-02-06:
@TracyCurry 

It sounds weird that it should send a global ACK when talking to a specific ECU.

Read the SENSE 42 document in the document folder. It shows the ACK protocol.

TracyCurry, 2025-02-07:
> [@TracyCurry](https://github.com/TracyCurry)
> 
> It sounds weird that it should send a global ACK when talking to a specific ECU.
> 
> Read the SENSE 42 document in the document folder. It shows the ACK protocol.

It sounds weird to me too although j1939-21 states its concern which is follow

[image]
And what i worry about is what if another ECU uses an global address as destination in ACK, does it still can communicate with ecu running this code correctly in specific case?

DanielMartensson, 2025-02-07:
@TracyCurry 

Can you give me som can traffic example.

TracyCurry, 2025-02-07:
> [@TracyCurry](https://github.com/TracyCurry)
> 
> Can you give me som can traffic example.

closed 2025-02-07 by TracyCurry

TracyCurry, 2025-02-07:
> [@TracyCurry](https://github.com/TracyCurry)
> 
> Can you give me som can traffic example.

I don't have a real platform yet. My project is still on early stage and I can only illustrate an scenario about my concern, for example, ecu a sends a request to ecu b, demanding an ACK, and ecu b replys back an ACK with a global address, will or should ecu a regards it as an valid ACK for its request? 
By the way, what really bother me is the ambiguity of the protocol and an definition being explained in more than one way just likes an itch

DanielMartensson, 2025-02-07:
@TracyCurry 

If you have any suggestions. Please send a PR.