DanielMartensson/Open-SAE-J1939 #64: How to clear the specific DM1 from another ECU without DM11?

ChunBoo, 2025-02-06:
Hi Daniel,

   As titled, "DM11 is a request for clearing DM1 messages. The reason why I don’t have implement it, is because in my opinion, it’s just an administrative burden that other ECU need to clear the DM1 error codes. I want the ECU it self to clear their own DM1 codes. The same way it activate the error codes by it self."  Could you please share the method of clearing specific DM1 from other ECUs without sending DM11 requests? Thanks.

Regards,
Bob.

DanielMartensson, 2025-02-06:
@ChunBoo 

I mean that the user itself should clear the DM1. But if you want to implement DM11 and clear DM1. Please write me sa PR.

ChunBoo, 2025-02-06:
> [@ChunBoo](https://github.com/ChunBoo)
> 
> I mean that the user itself should clear the DM1. But if you want to implement DM11 and clear DM1. Please write me sa PR.

Sure, Daniel. I would submit the PR for this feature later.

closed 2025-02-06 by ChunBoo

DanielMartensson, 2025-02-06:
> > [@ChunBoo](https://github.com/ChunBoo)
> > I mean that the user itself should clear the DM1. But if you want to implement DM11 and clear DM1. Please write me sa PR.
> 
> Sure, Daniel. I would submit the PR for this feature later.

@ChunBoo 

Look at DM3 how it is done. It could help.

ChunBoo, 2025-03-06:
@DanielMartensson  Hi, Daniel, I am sorry for the late response for this feature. I hope I can implement it at the end of this month.

DanielMartensson, 2025-03-06:
@ChunBoo 

That's OK.