DanielMartensson/Open-SAE-J1939 #66: STM32 Example

mustafaakkaya95, 2025-02-26:
Hello Daniel. Thank you for this work. I have stm32f407vet6 industrial development board and pcan-usb. I want to work using CubeIDE and PCANVIEW. First of all, I changed the "processor choice" option to "stm32" in the hardware.h file. Then I tried to run the code in the main.txt file. But there was no change after running the code. Is there a faulty or missing part in my code? What changes should I observe after sending a message via PCANVIEW?

/*
 * Main.c
 *
 *  Created on: 16 juli 2021
 *      Author: Daniel Mårtensson
 */

#include <stdio.h>
#include "Open_SAE_J1939/Open_SAE_J1939.h"
#include "ISO_11783/ISO_11783-7_Application_Layer/Application_Layer.h"
#include "stm32f4xx_hal.h"
#include "can.h"

void Callback_Function_Send(uint32_t ID, uint8_t DLC, uint8_t data[]) {
    CAN_TxHeaderTypeDef TxHeader;
    uint32_t TxMailbox;
    
    TxHeader.DLC = DLC;
    TxHeader.RTR = CAN_RTR_DATA;
    TxHeader.IDE = CAN_ID_EXT;
    TxHeader.ExtId = ID;

    if (HAL_CAN_AddTxMessage(&hcan1, &TxHeader, data, &TxMailbox) != HAL_OK) {
        // Gönderim hatası
        Error_Handler();
    }
}

void Callback_Function_Read(uint32_t* ID, uint8_t data[], bool* is_new_data) {
    CAN_RxHeaderTypeDef RxHeader;
    
    if (HAL_CAN_GetRxMessage(&hcan1, CAN_RX_FIFO0, &RxHeader, data) == HAL_OK) {
        *ID = RxHeader.ExtId;
        *is_new_data = true;
    } else {
        *is_new_data = false;
    }
}

void Callback_Function_Traffic(uint32_t ID, uint8_t DLC, uint8_t data[], bool is_TX) {
    printf("%s\t", is_TX ? "TX" : "RX");
    printf("%08X\t", ID);

    for (uint8_t i = 0; i < DLC; i++) {
        printf("%X\t", data[i]);
    }

    for (uint8_t i = DLC; i < 8U; i++) {
        printf("%X\t", 0U);
    }

    printf("\n");
}

void Callback_Function_Delay(uint8_t delay) {
    HAL_Delay(delay);
}

void SystemClock_Config(void) {
    // Sistem saat yapılandırmalarınızı buraya ekleyin
}

void CAN_Config(void) {
    CAN_FilterTypeDef sFilterConfig;

    // CAN1 yapılandırması
    hcan1.Instance = CAN1;
    hcan1.Init.Prescaler = 16;
    hcan1.Init.Mode = CAN_MODE_NORMAL;
    hcan1.Init.SyncJumpWidth = CAN_SJW_1TQ;
    hcan1.Init.TimeSeg1 = CAN_BS1_8TQ;
    hcan1.Init.TimeSeg2 = CAN_BS2_1TQ;
    hcan1.Init.TimeTriggeredMode = DISABLE;
    hcan1.Init.AutoBusOff = DISABLE;
    hcan1.Init.AutoWakeUp = DISABLE;
    hcan1.Init.AutoRetransmission = ENABLE;
    hcan1.Init.ReceiveFifoLocked = DISABLE;
    hcan1.Init.TransmitFifoPriority = DISABLE;
    if (HAL_CAN_Init(&hcan1) != HAL_OK) {
        Error_Handler();
    }

    // Filtre yapılandırması
    sFilterConfig.FilterBank = 0;
    sFilterConfig.FilterMode = CAN_FILTERMODE_IDMASK;
    sFilterConfig.FilterScale = CAN_FILTERSCALE_32BIT;
    sFilterConfig.FilterIdHigh = 0x0000;
    sFilterConfig.FilterIdLow = 0x0000;
    sFilterConfig.FilterMaskIdHigh = 0x0000;
    sFilterConfig.FilterMaskIdLow = 0x0000;
    sFilterConfig.FilterFIFOAssignment = CAN_FILTER_FIFO0;
    sFilterConfig.FilterActivation = ENABLE;
    if (HAL_CAN_ConfigFilter(&hcan1, &sFilterConfig) != HAL_OK) {
        Error_Handler();
    }

    // CAN geri çağırma işlevlerini ayarlama
    CAN_Set_Callback_Functions(Callback_Function_Send, Callback_Function_Read, Callback_Function_Traffic, Callback_Function_Delay);

    // CAN'i başlatma
    if (HAL_CAN_Start(&hcan1) != HAL_OK) {
        Error_Handler();
    }

    // CAN kesmelerini etkinleştirme
    if (HAL_CAN_ActivateNotification(&hcan1, CAN_IT_RX_FIFO0_MSG_PENDING | CAN_IT_TX_MAILBOX_EMPTY) != HAL_OK) {
        Error_Handler();
    }
}

int main(void) {
    // HAL ve sistem yapılandırmasını başlatma
    HAL_Init();
    SystemClock_Config();

    // CAN yapılandırmasını başlatma
    CAN_Config();

    // J1939 yapılandırmasını başlatma
    J1939 j1939 = {0};
    Open_SAE_J1939_Startup_ECU(&j1939);

    // SAE J1939 işlemi
    bool run = true;
    while (run) {
        // Gelen mesajları okuma
        Open_SAE_J1939_Listen_For_Messages(&j1939);

        // Uygulama kodunuzu buraya ekleyin
    }

    // ECU bilgilerini kaydetme
    Open_SAE_J1939_Closedown_ECU(&j1939);

    return 0;
}

DanielMartensson, 2025-02-26:
@mustafaakkaya95 

That is just a template as an example. Either you use the internal callback functions, or implement the hardware.h by your self.

Have a look at STM-PLC repository.

mustafaakkaya95, 2025-03-03:
I'm trying to try the ''Address Not Claimed'' example. I created two structures as in the code. I sent NAME from ECU 1 to all ECUs. The CAN message is sent according to the code in the ''SAE_J1939_Response_Request_Address_Claimed(&j1939_1);'' function. I can see this in PCANVIEW. But ECU cannot receive 2 messages. When I run it in Debug mode, in the "Open_SAE_J1939_Listen_For_Messages(&j1939_2);" function, it goes to return rx_msg; without entering the "if(is_new_message)" condition. What could be the error? Also, to activate the CAN message block, use "STM32_PLC_Start_CAN(hcan1, &j1939_1); ,
Which of the codes "STM32_PLC_Start_CAN(hcan1, &j1939_2);" should I call?  

/* USER CODE BEGIN Header */
/* USER CODE END Header */
/* Includes ------------------------------------------------------------------*/
#include "main.h"

#include "stm32f4xx_hal.h"

#include <stdio.h>
#include <stdlib.h>
#include <stdbool.h>
#include <string.h>

 /* Include Open SAE J1939 */
#include "Open_SAE_J1939.h"

/* Include ISO 11783 */
#include "Application_Layer.h"

#include "CAN_STM32.h"

/* Private includes ----------------------------------------------------------*/
/* USER CODE BEGIN Includes */
/* USER CODE END Includes */

/* Private typedef -----------------------------------------------------------*/
/* USER CODE BEGIN PTD */
/* USER CODE END PTD */

/* Private define ------------------------------------------------------------*/
/* USER CODE BEGIN PD */
/* USER CODE END PD */

/* Private macro -------------------------------------------------------------*/
/* USER CODE BEGIN PM */
/* USER CODE END PM */

/* Private variables ---------------------------------------------------------*/

	/* Create our J1939 structure with two ECU */
	J1939 j1939_1 = {0};
	J1939 j1939_2 = {0};

/* USER CODE BEGIN PV */

/* USER CODE END PV */

/* Private function prototypes -----------------------------------------------*/
void SystemClock_Config(void);
static void MX_GPIO_Init(void);
static void MX_CAN1_Init(void);
/* USER CODE BEGIN PFP */
/* USER CODE END PFP */

/* Private user code ---------------------------------------------------------*/
/* USER CODE BEGIN 0 */
/* USER CODE END 0 */

/**
  * @brief  The application entry point.
  * @retval int
  */
int main(void)
{

  /* USER CODE BEGIN 1 */
  /* USER CODE END 1 */

[... 178 lines cut ...]

/**
  * @brief  This function is executed in case of error occurrence.
  * @retval None
  */
void Error_Handler(void)
{
  /* USER CODE BEGIN Error_Handler_Debug */
  /* USER CODE END Error_Handler_Debug */
}

#ifdef  USE_FULL_ASSERT
/**
  * @brief  Reports the name of the source file and the source line number
  *         where the assert_param error has occurred.
  * @param  file: pointer to the source file name
  * @param  line: assert_param error line source number
  * @retval None
  */
void assert_failed(uint8_t *file, uint32_t line)
{
  /* USER CODE BEGIN 6 */
  /* USER CODE END 6 */
}
#endif /* USE_FULL_ASSERT */

DanielMartensson, 2025-03-03:
@mustafaakkaya95 
I cannot understand what you want to do?

Start with the main.txt example.

mustafaakkaya95, 2025-03-07:
In the "Address Not Claimed" example, at the exit "How many external ECUs are connected according to ECU 1? 0
How many external ECUs are connected according to ECU 2? 0
How many ECU cannot claim their address according to ECU 1? 1
How many ECU cannot claim their address according to ECU 2? 1"         
should I see?

DanielMartensson, 2025-03-07:
@mustafaakkaya95 

I don't understand what you want to do. 
It's very easy so implement Open SAE J1939 onto an embedded system. 

Do this first: Make sure you can transmit raw CAN messages between two controllers, before you implement Open SAE J1939

closed 2025-03-13 by mustafaakkaya95