DanielMartensson/Open-SAE-J1939 #75: Is the received from_other_ecu_tp_dt.data will be overwritten  when different ECUs broadcasting same types BAM?

ChunBoo, 2025-07-23:
Hi Daniel,

    When reading the TP.DT, [this](https://github.com/DanielMartensson/Open-SAE-J1939/blob/main/Src/SAE_J1939/SAE_J1939-21_Transport_Layer/Transport_Protocol_Data_Transfer.c#L25) line, it seems we are using the same array to store all incoming BAM packets, so in the case of we have two different ECUs that broadcasting the DM1 BAM at the same time, is ther a risk of data being overwritten? Thanks.

DanielMartensson, 2025-07-23:
@ChunBoo 
yes. If two ECU sends two BAM at the same time, only one message is going to be read.

It's a fix for that. Very easy. Use SA as an index for the struct.

For example:

Change this:

```
j1939->from_other_ecu_tp_dt.data[index7 + i-1] = data[i];
```

To this:

```
j1939->from_other_ecu_tp_dt[SA].data[index7 + i-1] = data[i];
```

The only problem why Is did not code for this is because if you having a struct array for 254 source addresses. Then the stack memory will be consumed.

But you can at least have 5-10 instead of 254. But you would need to make a queue then.

closed 2025-07-23 by ChunBoo

DanielMartensson, 2025-07-23:
@ChunBoo 
If you want, you can try this and send me a PR for that code? That would be very helpfull.